markqvist/Reticulum #1: LoRa does not support packet of 500 bytes

X-Ryl669, 2019-12-04:
As the title says, LoRa (at least Semtech's component) are limited to 256 bytes packet. Also, using such giant packets size takes a lot of air time to transmit and should be avoided since it'll decrease considerably the available duty cycle.

markqvist, 2019-12-11:
This is not an issue, but your personal opinion. While I do appreciate it, and find it interesting to discuss the implications of what you're saying here, I'm closing this.

As stated in the documentation, a 500 byte MTU is needed for Reticulum to function, primarily due to the length of RSA public keys. Such an MTU is perfectly achievable on Semtech LoRa hardware, or any other LoRa transceivers for that matter. If in doubt take a look at how RNode implements this :)

closed 2019-12-11 by markqvist

markqvist, 2021-09-02:
Although not "officially" supported, Reticulum can now, since commit cd8de6420155dc14f1b4743fc99d8c5bf68589cb, work with MTUs all the way down to 211 bytes. Bandwidth efficiency is negatively impacted by such a low MTU, but it does work.

The official default MTU is still 500 bytes, and running any other MTU than 500 will break intercommunication with networks that run the standard MTU.

So while I would not recommend using a different MTU, it is now possible if you want to :)

The change will be in beta 0.2.4, which I am going to release in a few days.