markqvist/Reticulum #3: #bug#: Is this a always true if clause

blademaster98, 2021-04-16:
In RNS.Identity.py file,  fuction:
    
     @staticmethod
    def validateAnnounce(packet):

    ...
    if len(packet.data) > Identity.DERKEYSIZE//8+20+Identity.KEYSIZE//8:
    ...
   
Since the last index of packet.data which has been stored into variable 'signature' is: [Identity.DERKEYSIZE//8+20+Identity.KEYSIZE//8], 
and array index in python starts at 0.
Is that the if clause shown above will always be true?
But fortunately, this seems no harm in the next steps , even if there is no app_data part in packet.data at all. The codes in if clause assign nothing to  variable 'signature'  since
[Identity.DERKEYSIZE//8+20+Identity.KEYSIZE//8:] equals to NULL.
   app_data = packet.data[Identity.DERKEYSIZE//8+20+Identity.KEYSIZE//8:]
But anyway I guess the expression of if is wrong,  the right one should be:
    if len(packet.data) > Identity.DERKEYSIZE//8+20+Identity.KEYSIZE//8 + 1:

Am I right?

blademaster98, 2021-04-16:
sorry  my bad. Not so good at python '[:]' operation.

closed 2021-04-16 by blademaster98

markqvist, 2021-04-22:
You got it ;)