markqvist/Reticulum #6: Communication Issues with Multiple Interfaces of Same Type

SePcGuy, 2021-07-26:
In a three node setup, with the first node being in subnet A, the second in subnet A & B, and the third being in subnet B, I cannot figure out a way to configure the second node that allows proper bidirectional communication with it having two UDP interfaces. Previously I found that having listen_ip as 0.0.0.0 for the default UDP interface and the forward_ip of whatever the broadcast address for the respective subnet of the broadcast interface would properly work and broadcast to the right network (on a multiple interface machine). After testing it specifically, specifying listen_ip to be the IP that the node was assigned on the subnet that should be transmitted on just breaks it entirely and no communication can be had. So, having listen_ip = 0.0.0.0 seemed to be the only way for it to work. 

However, here comes the problem I am having specifically... if you specify listen_ip as 0.0.0.0, you cannot have another UDP interface whatsoever, because it will give the following error message: "[2021-07-26 06:54:25] [Error] The contained exception was: [Errno 98] Address already in use". This happens if both the listen_ip is 0.0.0.0 or the assigned IP address on the node for the second subnet you wish to transmit on.

markqvist, 2021-07-26:
Hmm, this is a problem. Thanks for reporting. The way UDP interfaces are configured and set up at the socket level needs to be rethought a bit. I'll have a fix for this shortly.

SePcGuy, 2021-07-30:
Also another thing, the Filetransfer example doesn't seem to work. Screenshot of the error attached, I didn't figure it deserved it's own issue thread.
[image]

markqvist, 2021-08-19:
I'm working on a solution to the interface issue now, and should have a solution soon.

Regarding the file transfer example, I simply can't reproduce the issue. Could you post your python and Reticulum versions? I think I remember there was an encoding bug in that example at some point, but I believe I fixed some time ago.

[image]

markqvist, 2021-08-19:
Wait, I found the filetransfer exeample issue ;) working on it...

markqvist, 2021-08-19:
Okay, both issues should be solved now.

You can now define a UDP interface as follows:

```
[interfaces]  
  [[LAN UDP Interface]]
    type = UDPInterface
    interface_enabled = True
    outgoing = True
    device = eth0
    port = 4242
````

Or this would also work (note that both listen_ip and forward_ip is the _broadcast address_):
```
[interfaces]  
  [[LAN UDP Interface]]
    type = UDPInterface
    interface_enabled = True
    outgoing = True
    listen_ip = 10.0.0.255
    listen_port = 4242
    forward_ip = 10.0.0.255
    forward_port = 4242
````
I'd be interested in your results if you try out the latest version of the master branch (it's not in pip yet).

closed 2021-08-19 by markqvist

markqvist, 2021-08-19:
Also, version 0.2.2 breaks wire format compatibility with previous implementations, so you will have to update all instances!