markqvist/Reticulum #14: Porting to Android?

samuk, 2022-03-01:
This is probably silly, but now Rnode is running on generic ESP32, could Reticulum be ported to Android as an app? Would a USB or even Bluetooth connection to the ESP32 then work?

stevewa2066, 2022-03-02:
> This is probably silly, but now Rnode is running on generic ESP32, could Reticulum be ported to Android as an app? Would a USB or even Bluetooth connection to the ESP32 then work?
> 
> 
> 
> 

Where is the code for the ESP32?

I have the parts for doing a Bluetooth on the original Rnode. Just haven't had the time.

4c3e, 2022-03-02:
@stevewa2066 I was able to install rnode software onto my ESP32 board using https://github.com/markqvist/rnodeconfigutil

@samuk Nomadnet runs well on termux on Android ;)
For a native Android app, I ran into Kivy GUI library which would allow for an Android GUI app that could use RNS. The alternative could be to write a Kotlin Reticulum port which is something I'm increasingly interested in doing.

samuk, 2022-03-02:
Sorry I'm a bit of a noob here: So Nomadnet contains the entire Reticulum stack? All you need is an Android device and an ESP32 device to join the mesh & communicate?

Do you need root? Or could this be packaged up into a Play friendly app?

4c3e, 2022-03-02:
@samuk Nomadnet runs its own Reticulum instance if you aren't already running the rnsd utility in the background already. On android you can run Nomadnet using Termux which is a terminal emulator for Android. There are install steps in the Reticulum manual. I've only tried using the TCP transport from Android(TCP Client) -> Computer(TCP Server). LoRa should:tm: work by plugging in a LoRa board to your Android phone using a USB cabled, but I have not tested it because I don't have a cable that fits (yet).

markqvist, 2022-03-02:
I already started implementing Android support for Reticulum a while ago, and the entire stack is actually already functional on Android :) Due to differences in hardware access and drivers in Android, what is currently missing is a way to connect directly to devices such as RNodes and modems. I am working on this problem _right now_. It is not entirely trivial, so it needs to be done right to not be a pain later, but is is completely doable, and it will be ready very soon.

@samuk Yes and no. Nomadnet is easy to install on Android in Termux. See this part of the manual: https://markqvist.github.io/Reticulum/manual/gettingstartedfast.html#reticulum-on-android. After you have run "pip3 install rns", just run "pip3 install nomadnet", and you are ready to go. Currently you will need another device on your local network that hosts the LoRa interface. By default, the AutoInterface will be enabled out of the box on Android. If you add an RNode to another system on your network that also has the AutoInterface enabled, and enable Transport in that systems config, your Android device can also communicate over the LoRa interface. No root is needed. That was one of the original intentions of implementing Reticulum as a purely-userland stack, without any dependencies on kernel calls.

@4c3e Yes, Kivy works well for these purposes, and the associated py4a tools allows building Reticulum directly into Android APKs. It works very well, and **I already have working prototype APKs** with the Reticulum stack included, that so far runs well on all the Android devices I have tested them on.

My current priorities in these regards are: First, implementing the necessary hardware interfacing for Reticulum to connect directly to hardware devices on Android. Second, implement an Android app with messaging features (that is interoperable with the rest of the ecosystem of course).

samuk, 2022-03-07:
"Second, implement an Android app with messaging features (that is interoperable with the rest of the ecosystem of course)."

I wonder if it's worth forking https://briarproject.org/ they already have a nice UI worked out and 0.5million users. They might even accept a pull request upstream..

4c3e, 2022-03-07:
@samuk @markqvist I am an active Briar contributor 🙂 If there is any interest here, I'd be happy to make introductions and loop you all into the relevant channels. 
Briar project is currently doing their own independent mesh research, but I believe Reticulum could be a very useful transport to implement into Briar down the line.