markqvist/Reticulum #16: Transport Attached Sharable Links

4c3e, 2022-03-02:
One usability issue I've run into using Reticulum, is adding new Transports to my config in order to reach an RNS destination I want to send packets to. One way we could try and remedy this issue is by having RNS links that share Transport configuration. Once some kind of standard is realized, we could make RNS utilities that export current Reticulum "server" configs into a sharable link, and one that parses config links. 

For most of the transports, there are pretty straightforward ways to embed the config info into a URL and URL params. For LoRa radio settings, the "Data URL" spec might be useful here: https://developer.mozilla.org/en-US/docs/Web/HTTP/Basics_of_HTTP/Data_URIs

This format could then be combined with the Reticulum Internet Protocol URL scheme to provide shareable "Absolute" links that provide the RIP destination hash and the transport properties to reach that destination directly.

markqvist, 2022-03-02:
This is an interesting proposal. I think the current way you envision it is a little rough around the edges (I'm not sure the rip:// protocol specifier is the right thing here), but I get what you mean. It could be very useful.

As you mention it needs supporting utilities for import and export though, and I currently have my list of priorities quite packed full ;) For the 0.3.5 release though, I did plan on adding a "rnsconfig" utility that will make it easier to add and edit interfaces from both a text-ui and a graphical ui, so maybe it would be fitting to try to work this into these tools once we get to that point?

4c3e, 2022-03-02:
>I did plan on adding a "rnsconfig" utility that will make it easier to add and edit interfaces from both a text-ui and a graphical ui, so maybe it would be fitting to try to work this into these tools once we get to that point?

Sounds good, by the way, any recommendations for GUI frameworks for python?

markqvist, 2022-03-03:
Not really, yet. I will have to take another look around at the available options. I'd like to go with something easy to work with like PySimpleGUI, but in the sake of keeping dependencies at an absolute minimum, it might just have to be straight tkinter.