markqvist/Reticulum #17: Broadcast Example not working

HarlekinSimplex, 2022-03-11:
Hi Mark,
today I tried to get the Broadcast.py example up and running.
To achieve a proper test I did set up a test environment as follows:

DeltaMatrix  -> Unsigned.io
NexusServer 1 -> TestNode 1 ->  DeltaMatrix <- TestNode 2 <- NexusServer 2

All links are TCP/IP links and the arrows (Client -> Server) are indicating which node acts as server and which one acts as client.
DeltaMatrix is linked as a client to Unsigned.io (Reticulum TestServer provided by you) to give you the opportunity to access the Test Network.

The whole Testnetwork is alive for quite a while and works smoothly if used by nomadnet.
And as usual I checked those new NexusServer Nodes by running nomadnet on them as well.
All is fine. All anouncements are traversing the net as expected which I take as an indication that those nodes are fully functional.

Now I lauchend Broadcast.py on NexusServer 1 and NexusServer 2 with the following statement:
`python3 Broadcast.py --channel Public_Broadcast`

Output from NexusServer 1 was:
```
[2022-03-11 19:43:32] [Verbose] Configuration loaded from /root/.reticulum/config
[2022-03-11 19:43:32] [Verbose] Loaded 22 known destination from storage
[2022-03-11 19:43:32] [Verbose] Loaded Transport Identity from storage
[2022-03-11 19:43:32] [Notice] Broadcast example <d646696e18a31c104ccb> running, enter text and hit enter to broadcast (Ctrl-C to quit)
> [2022-03-11 19:43:42] [Verbose] Path to <39dc5820b09489c5e59f> is now 4 hops away via <fc0f105055cc1d4d5da9> on LocalInterface[37428]
[2022-03-11 19:43:50] [Verbose] Path to <5e3c89869609c35a8030> is now 4 hops away via <fc0f105055cc1d4d5da9> on LocalInterface[37428]
>
```

Output from NexusServer 2 was:
```
[2022-03-11 19:42:47] [Verbose] Configuration loaded from /root/.reticulum/config
[2022-03-11 19:42:47] [Verbose] Loaded 9 known destination from storage
[2022-03-11 19:42:47] [Verbose] Loaded Transport Identity from storage
[2022-03-11 19:42:47] [Notice] Broadcast example <d646696e18a31c104ccb> running, enter text and hit enter to broadcast (Ctrl-C to quit)
> [2022-03-11 19:43:22] [Verbose] Path to <39dc5820b09489c5e59f> is now 4 hops away via <4be6e824f5fabd4444d1> on LocalInterface[37428]
>
```

Typing someting on NexusServer 1 like ...
```
>Hello from Node 1
>
```

and NexusServer 2 like ...
```
>Hello from Node 2
>
```

Simply resolved in displaying another caret.
None of the messages entered showed up anywhere.

This behavior is independent from running nomadnet on those 2 NexusServer Nodes or not.
Nothing is received by both Broadcast.py scripts.

Can you pls adv what I'am missing here?

Best regards
Stephan

PS:
All those nodes indicated above are alive 24/7.
Actually the Broadcast Scripts are alive 24/7 as well - and will stay so at least until Monday to give you the opportunity to check them via the network with the IDs loged above.

HarlekinSimplex, 2022-03-12:
Some more informations on that topic:

Further tests showed that the Broadcast.py example does work nicely if you run it on one node, sharing the same instance.
However, as soon as I tried to use a remote (TCP) node as the link between the both broadcasters, nothing ever is revceived.
To check if have a general configuration issue here I used the Echo.py client server example. The results where as expected. Echo.py worked flawlessly in both directions. Still Broadcast.py still ceased to follow suite.

That raises a more severe question for me (or at least for our actual project):
Are Broadcast (aka unencrypted and public) messages are nor routed beyond the lokal node they were send from?

And if they should be routed like everything else, what is the caveat I still fell in all the time here?

Cheers
Stephan

HarlekinSimplex, 2022-03-13:
Today I tried to solve that problem by adding an announce to the Broadcast.py example like this:

```
    broadcast_destination = RNS.Destination(
        RNS.Identity(create_keys=True),
        RNS.Destination.IN,
        RNS.Destination.PLAIN,
        APP_NAME,
        "broadcast",
        channel
    )

    broadcast_destination.set_packet_callback(packet_callback)
    broadcast_destination.announce()
```

Technically that worked as expected until I tried to post a message.
The announce traversed the path beween both broadcast nodes es expected and could be heard all over the net as well.
Just sending a 'Hello World' caused a endless loop, that repeated the message sent endlessy between the nodes.
Adding another message from the second node, just added that second message to the endless rebroadcasted messages.
So the broadcast did get to its destination now, but was repeated endlessly.
Simply stopping the endpoints and restarting them did not stop the loop.
I had to stop the endpoints and the central node to break the chain.

After that I tried to change the mode of the endpoints from Transport=True to Transport=False.
That prevented the endless loop to occure all over the network, but did not help to prevent it between the broadcast endpoint node and its connected transport node. And again the endless loop could only be stopped by restarting the transport node that the broadcaster was connected to.

For me that looks like Reticlum has a larger problem with distributing broadcast messages without causing loops when it comes to routing PLAIN messages beyond link local. Maybe Reticulum does not prevent repeating a message to the network connection it got this same message from which of course would trigger such a endless loop. But that is just a wild guess.

Probably I miss something important still.
In that case help would be much appreciated here.

Next I'll try to get the GROUP type messsages working within our test net.
Used with a symetric key made public, this group message may be able to replace the broadcast PLAIN messages not working.
However, since routing those beyond link local listeners would require more or less the same routing algorithms and loop prevention Iike required for broadcasts, I fear that this message type will have the same issues like with broadcasts.
The difference would be simply the disabled encryption for PLAIN type messages.

And again, probably I miss something important here too, to make GROUP work as expected.
In that case help and/or advice would be much appreciated here too.

... to be continued as soon I#m done with the GROUP tests.

Cheers 
Stephan

markqvist, 2022-03-14:
I have just confirmed that the broadcast example is working as intended.

It seems you are expecting one or both of the following to be true:

- That undirected broadcasts will propagate over multiple hops via transport nodes. This is, by design, not the case. Broadcasts will only be heard by directly reachable endpoints. Anything else would be extremely inefficient, since global bandwidth cost for such a scheme rises exponentially with hops and is a fantastic DDoS vector.
- That SINGLE or GROUP destinations are globally adressable and/or routable. Again, this is not the case since it is not possible cryptohraphically to suppor this.

The "hack" you came up with of attaching and identity to a PLAIN destionation and broadcasting an announce was a pretty good idea ;) It is completely unintended behaviour though, and I missed that case in the filter rules. It should not be possible, since it can lead to routing loops, as you discovered ;) It will be fixed ASAP. Status on that is in #18. 

Reticulum _will_ feature globally routable GROUP destinations (ie., globally routable multicast) in the future, but it is currently not possible. You have to manage the endpoint transport via SINGLE destinations (or other methods) and ultimately pass that on to the GROUP destinations if they are not directly reachable.

Please post any questions, comments, doubts or ideas regarding this in the Discussion section.

Thanks for reporting your findings on this subject!

closed 2022-03-14 by markqvist