markqvist/Reticulum #24: [feature request] Add packets over TTN/Helium

jottr, 2022-03-31:
[TTN](https://www.thethingsnetwork.org/map) and [Helium](https://explorer.helium.com/) (to some degree) already provide coverage in many regions of the world.  
It could be useful to be use these LoRa-based networks to transmit Reticulum packets.

If routing Reticulum packets over other LoRa based networks is already possible a manual page/example would be useful.

4c3e, 2022-03-31:
> If routing Reticulum packets over other LoRa based networks is already possible a manual page/example would be useful.

This is already possible using Reticulum. Here's a guide for setting up encrypted LoRa messaging over Reticulum: https://unsigned.io/private-messaging-over-lora/. 

Please feel free to open a discussion if you have any other questions about setup.

markqvist, 2022-04-01:
As @4c3e mentions, the preferred (and currently only) way to use Reticulum over LoRa is using the methods already supplied and documented in various places. There is also information about setting up LoRa interfaces here:

- [Manual - RNode LoRa Interface](https://markqvist.github.io/Reticulum/manual/interfaces.html#rnode-lora-interface)
- [Installing RNode Firmware on T-Beam and LoRa32 Devices](https://unsigned.io/installing-rnode-firmware-on-t-beam-and-lora32-devices/)
- [How To Make Your Own RNodes](https://unsigned.io/how-to-make-your-own-rnodes/)

LoRaWAN networks (as opposed to raw LoRa, using an RNode for example) is generally not well suited for realtime bi-directional communication, since uplink and downlink packets can be spread out very far in time.

There are limited use cases where it might be useful to be able to "piggyback" on established LoRaWAN networks, for example for sending oppertunistic single-packet LXMF messages. This could make it possible to send very short (SMS-style) LXMF messages even when far away from any Reticulum LoRa nodes. This could be explored in the future.