markqvist/Reticulum #35: Adding support to other LoRa boards/device hats

Some-RandomProgrammer, 2022-04-04:
I saw that there was support for the author's RNode LoRa device, but would the same Interface work with other LoRa boards?
Like this one, for example: https://www.amazon.com/Frequency-Spectrum-Modulation-Raspberry-Interface/dp/B07VS47RQZ

Would I need to change any of the code from the RNode Interface to use this device?
I saw that the RNode uses a Semtech SX1276 Chip, but the example board uses a SX1262 Chip. Would that change the syntax needed?

Thanks for any replies.

4c3e, 2022-04-04:
Hello! @markqvist already has worked on a tool to configure off-the-shelf LoRa into functional RNodes: [rnodeconfigutil](https://github.com/markqvist/rnodeconfigutil).

It currently supports:

- The original RNode from unsigned.io
- Homebrew RNodes based on ATmega1284p boards
- Homebrew RNodes based on ATmega2560 boards
- Homebrew RNodes based on Adafruit Feather ESP32 boards
- Homebrew RNodes based on generic ESP32 boards
- LilyGO T-Beam v1.1 devices
- LilyGO LoRa32 v2.0 devices
- LilyGO LoRa32 v2.1 devices

I've also had success with using it with a Heltec ESP32 LoRa board.
Please feel free to follow up if you have any other questions.

markqvist, 2022-04-04:
Yes, the easiest route to support new LoRa transceiver boards is to add support to the [RNode Firmware](https://github.com/markqvist/rnode_firmware) and [RNode Config util](https://github.com/markqvist/rnodeconfigutil).

Adding support for the SX1262 family of devices is already on the roadmap, and work on that should start in the not-too-distant future. You can also have a look at [this section of the Reticulum repo README](https://github.com/markqvist/Reticulum#development-roadmap) for a general overview of where the development is heading.

Speaking in broad terms, there is huge potential for adding support for the many hundreds of kinds of transceiver modules available, and it is most definitely the intention to exploit that potential.

It needs to be done in the most optimally ordered, generalized and efficient way, so we don't end up needing to maintain 150+ codebases for different transceiver types ;) That is also why LoRa transceiver support currently exists under the RNode umbrella.