markqvist/Reticulum #39: anonymity?

burdges, 2022-04-05:
Is there any definition of anonymity for what's said in the readme?  Anonymity typically demands a global authoritative network map, ala Tor or mixnets, which mesh networks typically lack.

4c3e, 2022-04-05:
Great question! Destinations in Reticulum are simply hashes of public keys, so you could call destinations "pseudo-anonymous". Every packet is encrypted by default in Reticulum, except for announcement packets, which share destination public keys with the network, and packet types that have explicitly opted out of the public key encryption. The minimum amount of information you are leaking to the mesh network is metadata. This means information like packet timing and packet destinations. 

Currently, RNS supports "initiator anonymity" via it's [Link API](https://markqvist.github.io/Reticulum/manual/reference.html#link). What this means is that a node receiving an incoming link request, and all the peers between the initiator and the destination don't know where the request came from by default . Link's also use one-time ephemeral keys to provide forward secrecy. The RNS Link API is a fantastic tool to protect leaking metadata, and they can actually be used as a building block to create mixnets or tor/i2p type tunnels through the mesh network, if peers are willing to sacrifice significant amounts of bandwidth for initiator AND destination anonymity. I'm currently looking into ways to create multi-link tunnels with RNS in order to make mixnets possible for networks that want to protect metadata. Because announcements give nodes info about the connectivity of other nodes in the network, I think it might be possible to implement some form of onion routing on RNS (without "directory nodes") by creating multi-link paths between recently heard peers in the network. 

Let me know if you have any other questions, or clarifications.

burdges, 2022-04-05:
How do ACKs work then?

Attackers only need one direction typically.  It's plausible hiding sources benefits some specific cases, but afaik packets formats selling this feature never caught on, well normally anyone wanting this wants Tor or better.

You cannot build "tor/i2p type tunnels" without onion encryption, which handled MACs quite differently, use ciphers differently, etc.  It's all be useless if you tell each hop where your going though, hence some global network map.

I'll close this since I'm a happy with the answer that it's not anonymous, thanks.

closed 2022-04-05 by burdges

markqvist, 2022-04-05:
Reticulum is not a tool for achieving anonymity over networks that are already completely de-anonymized, such as the Internet.

It is a networking stack that allows you to build physical (and virtual) networks that can function anonymously to begin with.

You might be confusing the purpose of Reticulum with tools like Tor, which it definitely is not.

If you want to use Reticulum anonymously over the Internet, the current recommendation is to [use the I2PInterface support](https://markqvist.github.io/Reticulum/manual/gettingstartedfast.html#bridging-over-the-internet).

If you want to learn more about how acknowledgements work, or routing in general, you can have a look in the manual and the Python implementation.

I don't see any logic behind your remarks about attackers, anonymity not being present, and whether or not historic packet formats caught on, so I'm not going to address that

If there is something you'd like to discuss or explore in those regards, please be specific about it, and I'd love to discuss it.

Either way, glad you found out what you needed :)

4c3e, 2022-04-05:
Sorry for commenting on a closed ticket, just wanted to add a few points.

> You cannot build "tor/i2p type tunnels" without onion encryption, which handled MACs quite differently, use ciphers differently, etc. It's all be useless if you tell each hop where your going though, hence some global network map.

True! Onion encryption is possible with RNS, although it would not work exactly like tor's implementation of onion encryption. I do think circuits can be constructed in a Reticulum network by utilizing a node's local lists of destination announcements, removing the need for global directory nodes/network map. I'd be happy to go deeper into how mixnets could be built on RNS if you are interested at all. 

RNS is simply trying to be a efficient mesh networking system. Just like how applications like tor and i2p are built using TCP/IP, anonymizing networks can be built on RNS using Links. I don't think anonymous mesh routing is something that the base network layer should implement, especially on networking system with such low bandwidth requirements as RNS.

burdges, 2022-04-06:
> It is a networking stack that allows you to build physical (and virtual) networks that can function anonymously to begin with.

Anonymous physical layers are impossible under most adversary models.  Just fyi, I2P, Tor, and even Rogaway's OnionAE provide no anonymity under quite realistic models, like "global passive". 

> it would not work exactly like tor's implementation of onion encryption

It'll wind up broken.  I've not looked at your details, but you cannot build onion encryption from AEADs, which should be your link layer primitive.  In particular, packet size leaking route position is considered a break.

> I do think circuits can be constructed in a Reticulum network by utilizing a node's local lists of destination announcements

Your assuming an "honest but curious" adversary, which nobody takes seriously.

Anonymity has a long messy history, in which almost all designs wound up broken.

4c3e, 2022-04-06:
@burdges thanks for the feedback. Do you have any ideas to improve the RNS protocol, or approaches to take at the app-level to improve anonymity/build anonymous networks? Also, do you have any recommendations of projects that attempt to work under a "global passive" model?

burdges, 2022-04-06:
You should do your link layer encryption properly, likely Noise or tweaked TLS 1.3, but..  Avoiding MitM at the link layer already sucks without trusted CAs, not really sure honestly.  Are we better off leaking the topology to build some certificate infrastructure?  I donno..

Anonymity would usually be considered a problem for another layer.  I do think one could meaningfully discuss what and how metadata leaks against weak-ish but more nuanced adversaries, ala your omission of sender id, but analyzing this gets messy, requires thinking about what real adversaries want and why, and afaik nobody does this now.

As an example of a vaguely similar feeling mess, Tor bridges fight an ongoing battle with the Great Firewall, mostly funded by the US State Dept, but actually Tor never played to win because then the Great Firewall would catch up, and Tor would need to invent even more tricks.  Instead, Tor deploys their newest bridge techniques more slowly, in hopes that people who need them get them first.

It all kinda sucks honestly because we'll likely have way more mesh networks in the future, largely due to social collapse caused by climate change and ecosystem collapse.  It'd be nice if we understood more than we do about what this means for privacy..  Apologies for being depressing.

markqvist, 2022-04-06:
> I've not looked at your details

If you actually want to help this improve and build something, you are very welcome to do so by specifically pointing out *real* problems, and maybe even proposing *real* solutions. That requires a minimum of understanding though. Maybe you should start there.

If you just want to handwavingly throw around vague and fluffy intellectualia, please do it somewhere else, I don't have time for it.

> You should do your link layer encryption properly, likely Noise or tweaked TLS 1.3, but..

This is just a vague handwaing at Reticulums link-layer encryption not being "done properly". Your point assumes an honest but naively flawed implementation, without actually having read the implementation. Nobody takes that seriously.

> Avoiding MitM at the link layer already sucks without trusted CAs, not really sure honestly.

Please, demonstrate a MitM with Reticulum, and I will take you seriously.

>Are we better off leaking the topology to build some certificate infrastructure? I donno..

No, we are not. Doing away with the mess of certificate infrastructure is one of the points of Reticulum.
 
> It all kinda sucks honestly because we'll likely have way more mesh networks in the future, largely due to social collapse caused by climate change and ecosystem collapse. It'd be nice if we understood more than we do about what this means for privacy.. Apologies for being depressing.

I think you probably have a lot of valuable and interesting perspectives on these topics, and quite possibly also on the technical and cryptographic aspects of protocols like Reticulum. You are conducting yourself in a manner that is completely unconstructive though, and does not contribute antyhing. I don't have time for that.

If you just want to have these more superficial conversations about this and that related to cryptography and networking protocols, and want to just ad-hoc critique stuff you don't *really* understand, I can certainly see the fun and value in that, from a conversational perspective. But let's do so over a beer instead, this is not the place for it.

markqvist, 2022-04-06:
Transferring this to a discussion, since it is not an issue.