markqvist/Reticulum #44: Tor as a transport

darrenmartyn, 2022-04-08:
I probably can't implement this immediately, as I am still getting familiar with the software, but I noticed it already supports I2P tunnels as a transport layer.

It probably wouldn't be particularly hard to implement Tor support - it would just be a change to the tcpclient interface implementation to use a SOCKS proxy, and for the tcpserver interface implementation, just spinning up a hidden service (probably using the Stem library) pointing to the tcpserver port (bound to localhost by default in Tor mode?). 

Unless I am missing something?

Love the project by the way, I hope at some time to pick up some of the associated hardware.

4c3e, 2022-04-08:
>Unless I am missing something?

I don't think you are missing anything. Pretty sure right now you could use the existing TCPServer and TCPClient interfaces to tunnel a Reticulum connection through tor. 

Do you know if there are any python libraries that would provide an easy method of managing tor hidden services? Using I2P, you can tunnel normal Reticulum TCP connections, but there's also an [API](https://github.com/markqvist/Reticulum/tree/master/RNS/vendor/i2plib) that allows the Reticulum process to manage all the I2P-related things programmatically, and more efficiently that spawning a setting up a TCP interface for every I2P peer. 

One other fairly significant option that a tor interface could provide, is the ability for a RNS Instance to easily connect to some public-facing TCPServer, using tor exit nodes.

darrenmartyn, 2022-04-09:
Stem and PySocks would probably be the two "new dependancies" added to implement this. 

PySocks would allow TCPClient interfaces to connect over a SOCKS proxy to a public facing instance running a TCPServer interface, including ones operating as hidden services.

Stem will allow to programatically control the Tor process, for example, by launching a hidden service. 

> One other fairly significant option that a tor interface could provide, is the ability for a RNS Instance to easily connect to some public-facing TCPServer, using tor exit nodes.

That is pretty much it. And with hidden service support for TCPServer's, you can have location-hidden RNS Instances behind NAT that offer a "public facing" interface for clients to connect to. 

I think I'll try get a draft/test of adding SOCKS support to the TCPClient interface first. Should allow me to get familiar with the codebase.

markqvist, 2022-04-09:
That sounds like a good approach. Let me know if you bump into any problems or have questions along the way, I will try my best to be helpful :)