markqvist/Reticulum #55: Add Openwrt support

Vitaliy86, 2022-05-15:
Openwrt is good platform for integration.

markqvist, 2022-05-16:
Agreed. It has been a while since I have had anything to do with openwrt thoigh, so I need to get back up to speed there to produce anything useful.

If you have any further info on what kind of hardware you'd be using this with, and it what scenarios, I'd love to hear it.

whitecd, 2022-07-27:
Agree, oWRT would be a good platform, especially if there is additional support for the SX130x family of LoRA Gateway radios. These are currently implemented as mPCIe cards with SPI/USB interfaces.  LoRA gateway radios support concurrent 8x8 125KHz BW de-modulators and 1x 125/250/500KHz, fixed SF inter-Gateway back-haul de-modulator. Additionally, in the SX-1303 there is a 'listen-before-transmit' function to lessen contention issues.  Use of a LoRA gateway card would provide for significant frequency/SF diversity and another level of scalability to (urban) RNS networks.  Since the de-modulators are concurrent, gateway radios can support frequency hopping and/or dynamic LoRA channel assignments in crowded spectrum.
If implemented as a oWRT package, it should support the family of oWRT HW to include rPIs, etc.  Supporting LUCI would make configuration easier for the community at large.  Ideally, a Reticulum oWRT package that supports >1 RNodes  and/or 1xLoRA gateway radios would be ideal.  Besides the LUCI config, the novel code development would be defining a new interface  supporting the SX130x family of LoRA gateway radio(s).  The SX130x would look like 64+1 LoRA radios with up to any 10+1 acting as concurrent receivers.
Depending on scale, there are LoRAWAN gateways deployed on rPI HW. A step up in robustness/scale would be the Gateworks.com family of Ventana or Newport SBCs.  There are an emerging family of SX-130x LoRA gateway mPCIe boards available.  The RAK5146 (www.RAKwireless.com)  is a SX1303 board with Listen-before-Talk, and GPS.

markqvist, 2022-07-28:
Thanks for this informative overview @whitecd. In a few hours, I will be offline for the next 14 days, but will return to this once I am back!

markqvist, 2022-08-24:
This is officially on the roadmap now, but no guarantees on when I will get to it ;)

fab1an2, 2023-02-02:
Python is in Arduino and on Pico 2040. Meybe put reticullum to pico with wifi device and make a wifi mesh network independent from internet