markqvist/Reticulum #65: Using LoRa Interface throws occational errors

HarlekinSimplex, 2022-06-11:
Hi Mark,

actually I see some weird errors in my container log while using my ESP32 LillyGo T-Beams:

`nexus-radiobox  | [2022-06-11 16:54:15] [Error] A serial port error occurred, the contained exception was: index out of range
nexus-radiobox  | [2022-06-11 16:54:15] [Error] The interface RNodeInterface[RNode LoRa Interface] experienced an unrecoverable error and is now offline.
nexus-radiobox  | [2022-06-11 16:54:15] [Error] Reticulum will attempt to reconnect the interface periodically.
nexus-radiobox  | [2022-06-11 16:54:15] [Error] Error while transmitting on RNodeInterface[RNode LoRa Interface]. The contained exception was: an integer is required (got type NoneType)
`

The interface config I use looks like that:

`  [[RNode LoRa Interface]]
    type = RNodeInterface
    interface_enabled = True
    outgoing = True
    port = /dev/ttyRNODE
    frequency = 867200000
    bandwidth = 125000
    txpower = 7
    spreadingfactor = 8
    codingrate = 5
    id_callsign = <myCallSign>
    id_interval = 600
    network_name = <myNetworkName>
    passphrase = <myNetworkCluster>
    flow_control = False
`

Interessting is, that both netwoked PIs thrown this error exactly at the same point in time.
At least the log entries are always have identical timestamps.
I gues the laset line of the error message given above indicates the cause of the issue.
It looks like something shall have a default or needs to have assigned a value in the config.
But what?
This one did not show on my servers before the last update of RNS.
Since I updated to the most recent version and started using the beacon and name/password functionality this error startet to appear regardless if I run RNS within a dockerized environment or by running RNS on the host without docker.

Any idea or advice how to fix that would be appreciated.

Cheers
Stephan

markqvist, 2022-06-11:
Very interesting, thanks for reporting this. What interface is the other one throwing the error at the exact same time? That will help me narrowing it down.

markqvist, 2022-06-11:
Okay, never mind, I found it ;) This was a missing check in inbound processing for IFAC-enabled interfaces. If you want to try it out, you can grab the latest from master, or I can compile a .whl for you.

closed 2022-06-11 by markqvist

HarlekinSimplex, 2022-06-13:
It was simply a 1:1 LoRa link with identical interface specs. except the callsign extension, sitting in my shack aprox. 20cm apart.

HarlekinSimplex, 2022-06-13:
Well, since I have a fully automated build process setup with docker hub to create non-cached builds of my projects multiarch images every time I push something into the nexus git repository, just feeding in a hot fix is not to easy.

When do you think will that fix be included by just using ‘pip install rns’, which is exactly what my dockerfile does on building the nexus image?

markqvist, 2022-06-13:
Ah, it makes sense then. It is probably a packet coming from something else that is picked up by both receivers then.

It will be included in 0.3.8, which is more or less ready, except for some final tuning and tests. I will probably have it out during this coming week.

HarlekinSimplex, 2022-06-13:
Sounds good.

More tests have to wait until the end of next week since I’m off site anyway.

 

And yes it is a LoRa Network in my Office / Shack, that consists of 2 different Networks (Name/PW) with 3 nodes each linked together running RNS on 6 mixed RPi/OPi 32bit/64bit SBCs.

It is the LoRa side of my RNS test infrastructure to test and evaluate RNS, nomadnet and LXMF as transport layer for my Nexus Messenger Server and App.

So that makes up for 6 LoRa TRX having all the same radio configuration within just a couple of meters…

markqvist, 2022-06-13:
Haha, cool, sounds like a nice little lab. Keep an eye on the repo then, 0.3.8 should be out within the week.

commit 2685e043 (2022-06-11): Fixed missing check for zero-length packets on IFAC-enabled interfaces. Fixes #65.