markqvist/Reticulum #69: Data radio legality

Olliroxx, 2022-06-26:
In the [manual](https://markqvist.github.io/Reticulum/manual/whatis.html#where-can-reticulum-be-used), it lists `amateur radio digital modes` as a medium that reticulum can use. However, I'm fairly sure most countries (definitely the US and UK) ban the use of encryption on amateur bands. Is there some edge case in the law I'm not aware of? Did it mean to use amateur equipment in ISM bands (I don't know if that's legal either)? Is there something else I'm missing?

kc1awv, 2022-06-26:
_I am not a lawyer_, however in the United States, encryption is only allowed for command and control of amateur radio satellites. Is there a use case for that using Reticulum? Sure. But as it sits now, traffic using Reticulum to send and receive terrestrial data over US amateur radio would be prohibited, at least with my understanding of the Rules. Without getting into the politics of it all, I can foresee a future time where the encryption ban in the US is lifted.

The wording in the manual still is applicable for other countries, though. For example in Poland, it is perfectly fine to encrypt communications over amateur radio. I would suggest that the line in the manual should be updated to include the words "where allowed" or "where encryption on amateur radio is legal" to prevent confusion.

As an aside, encryption is fine on public service and commercial frequencies in the US, assuming you hold the proper license.

Olliroxx, 2022-06-26:
Thanks for the reply, what do you mean by public service frequency?

kc1awv, 2022-06-26:
> Thanks for the reply, what do you mean by public service frequency?
Fire, police, ambulance, road works, town/city/state government, schools...

Things not public service: CB, GMRS, FRS...

markqvist, 2022-06-27:
What @kc1awv says here is completely spot on:

> I believe the spirit of the manual is that it's up to the end user to be sure that what they are doing is legal where they live, and not rely on whether or not the manual of a program is giving them permission to do something that could be illegal.

A just marginally thorough coverage of what you can and cannot do when and where, when it comes to RF emissions would take up more pages than the manual itself, and varies wildly from area to area. Other countries that do not prohibit encryption on amateur radio spectrum include Germany and Denmark. Also, many commercial or personal spectrum license holders use amateur radio digital *modes* on commercially or personally licensed spectrum, since it is often the most efficient way to establish communication.

I am also not a lawyer, so take the following with a grain of salt, and as an interesting side-note. The following is my *personal* interpretation of the legal texts, and as such, just an opening for discussion on an interesting subject.

In general, all countries within the EU legislative framework and the United States follow the ITU Radio Regulations, which sets forth some guidelines for the actual legal implementation in each separate legislation. This is reflected for example in the United States by the _Title 47 CFR FCC Part 97_, which contains the actual rules governing the amateur radio service in the US. Specifically, [Part 97.113](https://www.ecfr.gov/current/title-47/chapter-I/subchapter-D/part-97/subpart-B/section-97.113) defines the "Prohibited Transmissions" for the amateur radio service. The exact section of concern reads as follows:

> Music using a phone emission except as specifically provided elsewhere in this section; communications intended to facilitate a criminal act; messages encoded for the purpose of obscuring their meaning, except as otherwise provided herein; obscene or indecent words or language; or false or deceptive messages, signals or identification.

Out of this section, it is obviously the *"messages encoded for the purpose of obscuring their meaning"* part that is relevant to this discussion. The point here is very clear. If you encode a message, in any way (encryption or not), *for the **purpose** of obscuring its meaning*, that will be a prohibited transmission. It is the *purpose* that is of importance here though. Encryption in and of itself is **not** illegal to use over amateur radio in the US, and in fact, Part 97 does not include the word "encryption" even once.

This means that there is plenty of situations where you can use systems that employ cryptography and encryption without breaking Part 97 rules.

Reticulum does not, per se, employ encryption to "obscure the meaning of messages". Reticulum *fundamentally needs* packets to be encrypted for the transport and routing layers to *even work at all*. If you could turn off the encryption and replace it with a dummy cipher, just inserting plaintexts, **nothing** would work. It would not be possible for the Transport layer to verify paths through the network, it would not be possible to sequence resource transfers, and it would not be possible to verify any sort of connectivity or data delivery to destinations.
 
As such, Reticulum first and foremost employs encryption because *it is the most efficient way to make a decentralised routing and mesh protocol work at all*, and frankly, it is *the only way* to do so. It is therefore solely the *intention* of the operator that bears relevance when using Reticulum over amateur radio spectrum, if you do not use Reticulum *because* you want to hide the content or meaning of messages, and nobody can legally prove otherwise, that part of the law cannot be used against you. A bunch of grumpy OMs who detest digital modes and anything more new-fangled than CW might get annoyed and confrontational though, and I frankly don't know what is worse, the FCC or the OM Police Department.

Even in the places where the above legal wording is implemented, there is plenty of situations where I would *personally* not have any qualms what so ever with using Reticulum, for example:

- Using nomadnet to access pages on an amateur radio node that is publicly accessible (there can be no intention of obscuring meaning, since the content being sent is accessible to all licensed amateurs).
- Using Reticulum to program a system for checking status of, and controlling a remote repeater site.
- Creating a home automation system allowing you to control stuff (again the purpose here is not to obscure the meaning, but to make sure others can not remotely turn on your furnace or unlock your gate)
- Sending messages to, and retrieving messages from a shared message board for amateurs.

I can also foresee a future where the legislation is changed, but it will not happen without it being challenged by users who want the change.

I take the stance that cryptography and encryption is already perfectly fine on amateur radio spectrum *in many cases*, but not all, and that we should work towards a situation where the prohibition of encryption is not just omitted, but the use of cryptography is actively protected and encouraged. This is my own stance, and it is definitely an unpopular one in many circles within the amateur radio community, mostly because of the lack of understanding of what cryptography actually is, and the value it provides.

I have yet to hear a good argument in favor of the prohibition of encryption within the amateur radio service.

Also, I will just move this to the discussion board.