markqvist/Reticulum #118: RNodeInterface[RNode LoRa Interface] hardware initialisation error (code 01)

herbsims, 2022-10-14:
I am running into a problem when I am trying to set up the Rnode Daemon.

[2022-10-13 13:24:59] [Notice] Opening serial port /dev/ttyACM0...
[2022-10-13 13:25:03] [Notice] Serial port /dev/ttyACM0 is now open
[2022-10-13 13:25:03] [Error] RNodeInterface[RNode LoRa Interface] hardware initialisation error (code 01)
[2022-10-13 13:25:03] [Error] A serial port error occurred, the contained exception was: Radio initialisation failure
[2022-10-13 13:25:03] [Error] The interface RNodeInterface[RNode LoRa Interface] experienced an unrecoverable error and is now offline.
[2022-10-13 13:25:03] [Error] Reticulum will attempt to reconnect the interface periodically.
[2022-10-13 13:25:03] [Error] Radio state mismatch
[2022-10-13 13:25:03] [Error] After configuring RNodeInterface[RNode LoRa Interface], the reported radio parameters did not match your configuration.
[2022-10-13 13:25:03] [Error] Make sure that your hardware actually supports the parameters specified in the configuration
[2022-10-13 13:25:03] [Error] Aborting RNode startup
[2022-10-13 13:25:03] [Error] Could not open serial port for interface RNodeInterface[RNode LoRa Interface]
[2022-10-13 13:25:03] [Error] The contained exception was: RNode interface did not pass configuration validation
[2022-10-13 13:25:03] [Error] Reticulum will attempt to bring up this interface periodically
[2022-10-13 13:25:03] [Notice] Started rnsd version 0.3.14

and I get the rapid flashing and have to reboot.  How can i tell what is wrong with my LoRa configuration?  I have changed the frequency to 900000000Hz as well (900 MHz).  I have verified that the LilyGo Chip is a SX1262 and shows 868/916.......

Here is what is in the config file.  

[[RNode LoRa Interface]]
  type = RNodeInterface

  # Enable interface if you want use it!
  interface_enabled = True

  # Serial port for the device
  port = /dev/ttyACM0

  # Set frequency to 867.2 MHz
  frequency = 867200000

  # Set LoRa bandwidth to 125 KHz
  bandwidth = 125000

  # Set TX power to 7 dBm (5 mW)
  txpower = 7

  # Select spreading factor 8. Valid
  # range is 7 through 12, with 7
  # being the fastest and 12 having
  # the longest range.
  spreadingfactor = 8

  # Select coding rate 5. Valid range
  # is 5 throough 8, with 5 being the
  # fastest, and 8 the longest range.
  codingrate = 5

  # You can configure the RNode to send
  # out identification on the channel with
  # a set interval by configuring the
  # following two parameters.
  # id_callsign = MYCALL-0
  # id_interval = 600

  # For certain homebrew RNode interfaces
  # with low amounts of RAM, using packet
  # flow control can be useful. By default
  # it is disabled.
  flow_control = False

markqvist, 2022-10-14:
According to the error code in the log output, the radio module could not be turned on. This *might* be a problem with the supplied parameters, but they look fine.

I think what is most likely here is that the incorrect firmware was flashed to the device, and the firmware cannot communicate with the LoRa module.

Can you let me know exactly what device you are using, and exactly what choices you made in the firmware installer?

markqvist, 2022-10-14:
Just to clarify, the error codes in the log output and the rapid blinking indicates that there is a hardware failure when trying to start up the radio module. This could be because the device is simply broken, but I really doubt that. As I wrote above, I think it is most likely just because it doesn't have the correct firmware installed :)