markqvist/Reticulum #125: Adding LoRa config causes "Android-specific interface was used on non-Android OS"

Vault2501, 2022-10-21:
Hi, I'm currently running 0.3.16, and as soon as I add the LoRa config to the reticulum config file, I get this upon starting nomadnet or sideband:

```
nomad@rpi4b:~/.reticulum$ nomadnet 
[2022-10-21 20:21:36] [Error] The interface "RNode LoRa Interface" could not be created. Check your configuration file for errors!
[2022-10-21 20:21:36] [Error] The contained exception was: Android-specific interface was used on non-Android OS
```

My LoRa config snippet:
```
 [[RNode LoRa Interface]]
    type = RNodeInterface
    interface_enabled = True 
    outgoing = true
    port = /dev/ttyACM0
    frequency = 867200000
    bandwidth = 125000
    txpower = 7
    spreadingfactor = 8
    codingrate = 5
    # id_callsign = MYCALL-0
    # id_interval = 600
    flow_control = False
```

With the auto generated config, nomadnet and sideband start as expected.

markqvist, 2022-10-21:
Confirmed, I will have a fix out shortly.

markqvist, 2022-10-21:
I am not in reach of my computer right now, but as soon as I am, I will package this fix into a release and upload it. In the meantime, you can manually apply the very simple patch in https://github.com/markqvist/Reticulum/commit/ed4070a3d14f0976b0c79b63b1819e02a1f48c55.

closed 2022-10-21 by markqvist

markqvist, 2022-10-21:
It is released now as version `0.3.17`. Thanks a lot for reporting this!

commit ed4070a3 (2022-10-21): Removed stray import. Fixes #125.