markqvist/Reticulum #149: Error in the documention for the wire format

markqvist, 2022-11-06:
### Discussed in https://github.com/markqvist/Reticulum/discussions/148

<div type='discussions-op-text'>

<sup>Originally posted by **thatv** November  5, 2022</sup>
Hi there, currently i am looking at the doc of the project and trying to understand the wire format. I am not sure if i get something wrong or I have found an error in the section with the Packet Examples. The second and third example stating a Hop count of 00000111 which is stated as zero hops. shouldn't it stated as 7 hops?

+- Packet Example -+

   HEADER FIELD   DESTINATION FIELD   CONTEXT FIELD  DATA FIELD
 _______|_______   _______|_______   ________|______   __|_
|               | |               | |               | |    |
00000000 00000111 [HASH1, 16 bytes] [CONTEXT, 1 byte] [DATA]
|| | | |    |
|| | | |    +-- Hops             = 0
|| | | +------- Packet Type      = DATA
|| | +--------- Destination Type = SINGLE
|| +----------- Propagation Type = BROADCAST
|+------------- Header Type      = HEADER_1 (two byte header, one address field)
+-------------- Access Codes     = DISABLED

Thanks for the Clarification. And a nice Project btw.</div>

thatv, 2022-11-12:
create PR to fix this issue

markqvist, 2022-11-14:
Thank you very much!

markqvist, 2022-11-17:
Fixed by PR.

closed 2022-11-17 by markqvist