markqvist/Reticulum #175: Any interest in adding static code analysis?

mkinney, 2022-12-17:
If you run: `pylint RNS | wc -l` you get 11,109 lines of warnings. Most are trivial to fix. There are some that may take some effort. Other warnings could be ignored with settings.

I can make PRs that would bring the code into compliance, if that is desired.

I done efforts like this in the past. They make the code more consistent and sometimes even find latent bugs.

markqvist, 2022-12-17:
While I appreciate the good intentions to improve code readability, I am not sure this is the right time. The proposed PR will create quite a bit of extra work for me to clean up as it is right now, and it bears no real functional improvement or significant readability improvement to the code.

As an example of problems that can arise with automated code improvement tools like pylint:

- The added docstrings will now, most probably, mess up the automated Sphinx builds of the API documentation, cluttering it with unneccessary "documentation" for classes or API functions that are irrelevant to people using the documentation. This is a net negative, and needs to be cleaned up, and taken care of properly in the documentation build system. This takes time from other efforts that I currently prioritise higher.
- I don't know the rationale behind why `pylint` complains about having a method named `exit()` on a class, but renaming it to the arbitrarily different `exit_now()` does not really improve anything from a readability standpoint, and now breaks anything else that might call that function (in other apps using RNS). I don't actually *know* if any of the apps I maintain call that function specifically, but I will of course have to go and check everything, which again takes time.

Those two are obviously relatively contained and simple examples, but when you let `pylint` loose on the entire codebase, I am very certain that things will break in more ways than one, simply because some sublety is overlooked, or because `pylint` suggests something that *actually* doesn't make sense.

Trawling through all of that and finding potential errors and debugging them will, most likely, be a more or less massive undertaking, and I think the net benefit in no way justifies that right now, especially seeing that this is an almost purely subjective and stylistic change, with no real functional benefits.

There is still a lot of things that I need to document, comment properly, and restructure for legibility in the entire codebase, along with updating the actual documentation and manual to account for changes and improvements. This  the primary focus I am moving towards at the moment. Before that work has been carried out, I don't think it makes much sense to run automated code-quality checkers on the repository. It will only hamper my ability to effectivelt and concisely carry out that work.

I hope I don't come off as negative about the initiative, because I am not. I simply don't think this is the right time to do it, for the reasons above, and the fact that my current ability to oversee the process is quite limited due having other work on the project that I prioritise higher for now.

I hope you will consider helping with this effort a bit later in the process, when the current phase of documentation improvement is finished.

mkinney, 2022-12-17:
Ok. I'll close the PR.

closed 2022-12-17 by mkinney