markqvist/Reticulum #180: Consider Crystals Kyber and Crystals Dilithium, Upgrade to AES-256, upgrade to SHA3-512

Destroyinator69420, 2023-01-12:
You are not using the strongest version of the encryption algorithms available. For instance, you are using AES-128, which can be weakened using a quantum computer. The simplest method to combat this would be to use AES-256. You are using SHA-256, although I do not know whether you are using SHA2 or SHA3. If you are using SHA2, then you should be using SHA3, which is more recent and widely regarded as more secure. SHA3 supports an option for a 512 bit hash, this is widely regarded as more secure than SHA256 in regards to collision attacks. I would also recommend integrating Crystals Kyber into the asymmetrical cryptography, as doing so will make your network quantum resistant, but the Crystals developers recommend using Crystals Kyber with an already established algorithm. Irregardless of whether you encrypt with Elliptic Curve Diffie Helman or Crystals Kyber first is of no consequence so long as the keys used to encrypt them are generated separately. You should also consider using Crystals Dilithium as a hashing algorithm in addition to SHA3-512. This will provide two layers of security in terms of message authentication. I would recommend Dilithium5, although there is a faster version that uses AES in counter mode instead of shake to expand the polynomials, it is called dilithium5-AES. It uses AES to take advantage of hardware accelerated AES when possible, although I would stick with Dilithium5 becuase SHAKE is an irreversible function, unlike AES. There is an open source library called liboqs created by the Open Quantum Safe project. It has a python wrapper and is written in C. All of the algorithms I have mentioned in this issue ticket are available in OpenSSL if not in liboqs.

Other than my crituqe of your cryptography choices I would like to congratulate you for fighting the good fight and proving that a more decentralized and secure internet is possible. I would also like to congratulate you on not selling out to the Web3 craze. Hopefully reticulum gets the attention it deserves.

markqvist, 2023-01-12:
Hi @Destroyinator69420

Thanks for your comments and critiques.  I am of course very well aware of all of the primitives that you mention.

### TL;DR

I really don't mean this as a personal insult, but I will have to say it clearly from the start: Your recommendation makes no sense, and your understanding about cryptographic systems seem shallow at best. I hope you will follow your interest in the field, and work to deepen your understanding.

There are specific, and very well-considered reasons for the choice of cryptographic primitives in Reticulum, with the goal of bringing the highest degree of real-world security and functionality to the people using the stack.

The composition of a cryptographic primitive suite has **far** wider consequences than just the Ultra-Security-Hype-Factor-Buzzword-Score™ that it can generate, but that seems to be what you are mostly interested in here.

I am concerned with *real* security and reliability, for *real-world* applications.

### Understanding Post-Quantum-Cryptography

It seems like there is a tendency for people to read about the Crystals project and think that we now have "the solution to quantum-resistant cryptography", and that we should all move to this immediately to be safe. This is complete a misconception.

The *Cryptographic Suite for Algebraic Lattices*, also known as "Crystals", is currently a *research project*, intended as a contender in a standardised suite of cryptographic primitives, that will let us continue to deploy secure cryptographic systems, in a ***hypothetical*** world, where highly efficient quantum computers have become a reality.

It is **not** a well-audited, thouroughly proven set of cryptographic tools. The NIST PQC project has not even completed *selecting* candidates for standardization into a PQC suite of cryptographic primitives.

If I had to place my bets, they would be that at least some of the primitives you mention will get selected as final candidates in the NIST PQC project. After that should then come *years* of scrutiny, review and testing by competent cryptographers and mathematicians other than the authors, before **any** serious use is considered.

People are working seriously on these projects already, because there is a *potential risk* of them being necessary in the future, and because it will take a significant number of years to complete the task. Should this *potential risk* turn into reality, we want to be *prepared*, and therefore it is something we need to work on now.

Does that mean that everyone should quickly jump boat to random subsets of contenders for PCQ standard primitives?

**No**, absolutely not.

### Does Unicorns Haunt You?

You have probably read a lot of alarmist "articles" from various "news" outlets about the impeding quantum-computer-cryptography-armageddon. Such "articles" are fantastic attention-grabbers. They are pretty far off from reality, though.

As far as we currently know, it *might* be possible to build a quantum computer of the required complexity and speed needed to, amongst other things, efficiently break elliptic curve cryptography, and to significantly weaken symmetric ciphers like AES. There is a lot of *ifs* here though:

- We don't know *if* a quantum computer of the required complexity is actually possible to build.
- We don't know *if* the economic and material resources exist to allow the construction and operation of such a computer.
- If it is even possible to build a quantum computer with several million qubits and sufficiently low gate error figures, we have no idea how long it will take us to realise this. It may be fifty years, it may be five thousand. It is most certainly not something that is right around the corner.
- We don't know if the costs and other requirements involved in operating such a computer are feasible for running tasks required for breaking cryptography.

Said in another way, we *think* that we *might*, at some indeterminate point in the future, *possibly* possess a device that, in *theory*, has a *hypothetical* way of breaking *some* parts of our current cryptography systems.

That is enough to get cryptographers working on potential solutions, and is the most reasonable course of action. But anything much more than that is much closer to magical thinking, than rational risk asessment.

For most people, actually worrying about this is akin to being afraid that a unicorn might suddenly, and in broad daylight, materialise out of thin air, and proceed to violently mow you down with it's glittering horn, in a shower of rainbow-coloured sparkles.

Literally every person working in the field of cryptographic systems are perfectly well aware of developments in both quantum computing and quantum-resistant cryptography. I can assure you that nobody will be taken by surprise here, and that *if* the time comes, we will all be ready, collectively.

This is of course also true for Reticulum, which is designed to be completely modular in terms of primitive suites, and can have all parts changed with just a few lines of code.

But implementing any of your suggestions right now would literally and directly harm the reliability, functionality, usability over low-bandwidth mediums, universality, ease of use, and not least the actual security of Reticulum.

If you think you know otherwise, please outline the physical bounds for a quantum-compute capable system that can efficiently break X.25519, Ed25519, AES-128 and SHA-256 as used in Reticulum, in a time frame that is just *marginally* useful, let's say 100 years.

markqvist, 2023-01-12:
Some further reading for anyone that might be interested:

- https://en.wikipedia.org/wiki/Shor%27s_algorithm
- https://security.stackexchange.com/questions/48022/what-kinds-of-encryption-are-not-breakable-via-quantum-computers/48027#48027
- https://www.schneier.com/blog/archives/2022/02/breaking-245-bit-elliptic-curve-encryption-with-a-quantum-computer.html
- https://avs.scitation.org/doi/10.1116/5.0073075

markqvist, 2023-01-12:
And just another note to @Destroyinator69420 specifically; since you seem fond enough of Kyber to recommend it for actual use in projects, you'd probably be interested in noting this: https://groups.google.com/a/list.nist.gov/g/pqc-forum/c/Fm4cDfsx65s/m/0dBVXOcSCAAJ