markqvist/Reticulum #204: Feature Request: pass link id to response generator

acehoss, 2023-02-06:
I was working on a request/response application that maintains a session state for the lifetime of the Link. When the requester identifies on the link, a session is started and this session state lasts until the link is torn down.

Digging through the source code, I didn't find any way to get back to the link from the response generator callback. 
```
response_generator(path, data, request_id, remote_identity, requested_at)
```
It would be possible to find the correct link by identity, but this limits the application to one link per identity (which wouldn't work for this application). I realize it would be a breaking API change, but adding the `link_id` to the parameter list would resolve the problem.

As a workaround, I used a big hack and wrapped the `RNS.Link.handle_request` function:
```
_link_handle_request_orig = RNS.Link.handle_request

def link_handle_request(self, request_id, unpacked_request):
    global sessions
    for session in sessions:
        if session.link.link_id == self.link_id:
            RNS.log("Associating packet to session", RNS.LOG_DEBUG)
            session.next_request_id = request_id
    _link_handle_request_orig(self, request_id, unpacked_request)

RNS.Link.handle_request = link_handle_request
```

markqvist, 2023-02-09:
Hmm, this *should* actually be possible, and I could've sworn I already implemented it... Let me just have a look :)

markqvist, 2023-02-09:
Thanks for bringing this up @acehoss. It's added now in fc818f00f1b34a563390f362dd83b7c56b01b1c9, and the callback signature has also been updated in the [API documentation](https://markqvist.github.io/Reticulum/manual/reference.html#RNS.Destination.register_request_handler).

It'll be in the next release, but you can start using if you just import RNS directly from a clone of this repo or just `make debug` your own temporary wheels.

markqvist, 2023-02-09:
Just as a note, the old signature (without link_id) is still possible to use, as to not break backwards compatibility, but *may* go away in the future. For now I'll just leave it in though.

closed 2023-02-09 by markqvist

acehoss, 2023-02-09:
Awesome, thanks!

markqvist, 2023-02-09:
Let me know if it works out ok for your purposes or not!

acehoss, 2023-02-10:
Works great, thanks!