markqvist/Reticulum #223: Request/Response does not support bytes objects

acehoss, 2023-02-10:
If I try to send binary data in a request or response, usmgpack complains:

```
RNS.vendor.umsgpack.UnsupportedTypeException: unsupported type: <class 'bytearray'>
```

I'm working around this for now by base64 encoding the data.

markqvist, 2023-02-11:
That is odd. That *should* work. If it doesn't I'll have it fixed in a hurry :)

acehoss, 2023-02-13:
Currently my response generator returns an array with several elements, at least one of them being a `bytearray` object. 

The exception is raised at [Link.py:580](https://github.com/markqvist/Reticulum/blob/3706769c337c8f37eae11c9bea33269416967f5d/RNS/Link.py#L580):
```
packed_response = umsgpack.packb([request_id, response])
````

acehoss, 2023-02-14:
Found the right solution here: umsgpack doesn't support `bytearray` objects, but it _does_ support `bytes` objects. Converting the `bytearray` to `bytes` before sending fixed the problem.

closed 2023-02-14 by acehoss

markqvist, 2023-02-14:
I couldn't for the life of me reproduce this, until I remembered the specifics of *bytes* vs *bytearray* classes in Python. This happens because the `umsgpack` serialiser doesn't support the mutable *bytearray* type. It does support the *bytes* type however, so you can do this:

```
ba = bytearray([0x55, 0x45, 0xff])
b = bytes(ba)
```

markqvist, 2023-02-14:
Haha, you beat me to it by about 15 seconds! Apparently we were looking at this exactly syncronously!