markqvist/Reticulum #230: 802.11ah / HaLow

cryptoquick, 2023-02-19:
Does this support [802.11ah](https://en.wikipedia.org/wiki/IEEE_802.11ah)? It might be cool to use this with a device [like this](https://www.crowdsupply.com/teledatics-inc/halo-td-xpah).

4c3e, 2023-02-20:
I have one of those devices I've been meaning to test with. I *think* you should be able to set it up in a configuration where it will work with no work needed on the RNS side. Still need to sit down and experiment with it though.

markqvist, 2023-02-22:
Yes, since the HaLow devices will pass normal ethernet frames, the `AutoInterface` should work fine with them. There *may* be quirks that I don't know about, since I currently don't have access to any HaLow hardware.

markqvist, 2023-02-22:
If anybody wants to donate the cost of four of the TD-XPAH, or the units themselves, I can make sure they are supported.

cryptoquick, 2023-02-23:
> If anybody wants to donate the cost of four of the TD-XPAH, or the units themselves, I can make sure they are supported.

Do you accept Bitcoin? If so, post an address, I might be able to get a few of my freedom-minded friends to join me in financing this. Would $500 worth be enough?

cryptoquick, 2023-02-23:
Found your donation address, I just sent 500k sats, trying to get some friends to do the same!

Sent to:
`3CPmacGm34qYvR6XWLVEJmi2aNe3PZqUuq`

4c3e, 2023-02-25:
I can donate the cost of one HaLow device.

markqvist, 2023-02-26:
Thank you so much everyone! Enough donations have come in now to cover the cost of the units! I will be ordering some in due course :)

pizzadoglover911, 2023-04-02:
How does 802.11ah differ from current 'mesh' or long range wifi solutions? 

I see they note it can support more clients. Of course higher bandwidth than Lora. I read it is in ism band (915mhz) yet also mentions 5 and 6 ghz. Also their comparison chart lists Lora as less secure?

Looks like a neat bit of hardware, though I am not sure what purpose it serves.

4c3e, 2023-04-04:
It's the first Wifi hardware that I know of that operates in the sub-GHz range. I'm not an expert on radio, but my understanding is that these bands are able to physically propagate further than the 2.4 and 5GHz bands. 

The halow boards seem to offer a middle ground between LoRa and traditional Wifi. The hope is that it can offer 2-3km range at a higher bandwidth than LoRa. Also, because the halow boards are built to work with the traditional IP stack, there's a chance it will integrate with more traditional tools nicer, although that aspect isn't totally relevant for RNS.

markqvist, 2023-04-05:
I have just gotten my hands on the HaLow boards donated. I won't have time to work with them properly until in a couple of weeks, but I'll report on the progress as soon as I know something.

cryptoquick, 2023-04-05:
Awesome! And to answer @pizzadoglover911, these boards support both the 802.11ah HaLow and 802.11s mesh networking standards, but I'm not sure how 802.11s compares to proprietary meshnet backhauls used by OEMs, nor am I sure how widely used that is. But it could possibly pair well with Reticulum. I look forward to seeing what can be done!

The-One-Law, 2023-04-06:
sub-GHz is pure gold if you are trying to build a mesh through forest (or jungle).
We did that with ISM freqs (400mhz) but the data rates were only enough for simple tracking.

og1en, 2024-02-17:
I need 802.11s mesh