markqvist/Reticulum #240: peering hash mismatch NetBSD -> macOS

gdt, 2023-02-23:
Starting rnsd on macOS and NetBSD, I get "authentication has was incorrect" on the mac, and ok received hash on NetBSD (via an extra log statement).

It seems fairly obvious that the problem must be on the NetBSD transmitter, given that the mac works with sideband/Android and presumably Linux, and I'm guessing it's about the bytes stuck into fe80 that denote interface number being used in the hash, when they don't appear on the wire -- but so far that's a guess.

gdt, 2023-02-23:
Also, there is a hash mismatch NetBSD to NetBSD.   This is consistent with a bug in NetBSD TX.

gdt, 2023-02-23:
When sending, the address is `fe80:1::` but on receipt it is `fe80::`.  This is the BSD convention of sticking the ifindex into the LLaddr so that it can be used with recvfrom/sendto without binding the address.  So we need to remove that.   Interesting that this issue doesn't show up on macOS.

gdt, 2023-02-23:
Kludging around the address, it works.  Will figure out a real fix and submit a PR.

markqvist, 2023-02-24:
Aha, that is interesting! Had no idea that NetBSD uses that convention. Thanks for working on this!

gdt, 2023-02-24:
I actually find it surprising that it isn't showing up on other systems.  I think this is from KAME, which was one of the earliest IPv6 implementations in the 90s.   I am curious if there are reports of success on FreeBSD, OpenBSD or illumos, basically anything but Windows/Linux/Mac

Erethon, 2023-02-25:
> I am curious if there are reports of success on FreeBSD, OpenBSD or illumos, basically anything but Windows/Linux/Mac

OpenBSD 7.2 currently fails with:
```
[2023-02-25 00:13:29] [Verbose] No valid Transport Identity in storage, creating...
Traceback (most recent call last):
  File "/home/dgrig/.local/bin/rnsd", line 8, in <module>
    sys.exit(main())
  File "/home/dgrig/.local/lib/python3.10/site-packages/RNS/Utilities/rnsd.py", line 71, in main
    program_setup(configdir = configarg, verbosity=args.verbose, quietness=args.quiet, service=args.service)
  File "/home/dgrig/.local/lib/python3.10/site-packages/RNS/Utilities/rnsd.py", line 41, in program_setup
    reticulum = RNS.Reticulum(configdir=configdir, loglevel=targetloglevel, logdest=targetlogdest)
  File "/home/dgrig/.local/lib/python3.10/site-packages/RNS/Reticulum.py", line 259, in __init__
    RNS.Transport.start(self)
  File "/home/dgrig/.local/lib/python3.10/site-packages/RNS/Transport.py", line 145, in start
    Transport.identity = RNS.Identity()
  File "/home/dgrig/.local/lib/python3.10/site-packages/RNS/Identity.py", line 365, in __init__
    self.create_keys()
  File "/home/dgrig/.local/lib/python3.10/site-packages/RNS/Identity.py", line 368, in create_keys
    self.prv           = X25519PrivateKey.generate()
  File "/home/dgrig/.local/lib/python3.10/site-packages/RNS/Cryptography/Proxies.py", line 14, in generate
    return cls(X25519PrivateKey.generate())
  File "/home/dgrig/.local/lib/python3.10/site-packages/cryptography/hazmat/primitives/asymmetric/x25519.py", line 42, in generate
    raise UnsupportedAlgorithm(
cryptography.exceptions.UnsupportedAlgorithm: X25519 is not supported by this version of OpenSSL.
```

which is because OpenBSD ships with LibreSSL and not OpenSSL. Having said that, LibreSSL 3.7.0 that was released ~2 months ago will have X25519 support according [to the release notes](https://marc.info/?l=openbsd-announce&m=167085276320510&w=2). 

I haven't (yet) tried compiling python with openssl support to see if it works or not.

Erethon, 2023-02-27:
> I am curious if there are reports of success on FreeBSD, OpenBSD or illumos, basically anything but Windows/Linux/Mac

I spent some more time testing this. I had to edit Cryptography/Provider.py` to force RNS to use the built-in encryption primitives instead of using the LibreSSL in Python as a temporary workaround.

I've setup an OpenBSD 7.2 node in testnet (address `6c6b3102c864fc69f186d550dbdd3f5e`) and it works fine. I also spun up an **x86** OpenBSD 7.1 node and it was able to access the 7.2 node on the testnet just fine. Unfortunately, the two nodes aren't on the same network, so I can't test local ipv6 connectivity. I can however spin up two local OpenBSD VMs and test it out if you think it's gonna be helpful.

@markqvist I'll leave the `6c6b3102c864fc69f186d550dbdd3f5e` node online in the testnet for now, feel free to use it for experiments if you wish, since it uses the custom python encryption implementation.

markqvist, 2023-02-27:
> I spent some more time testing this. I had to edit Cryptography/Provider.py to force RNS to use the built-in encryption primitives instead of using the LibreSSL in Python as a temporary workaround.

That's strange. Does LibreSSL not support X25519? Or does it just have a different API? I really wish we had a bare-minimum, well-tested, audited primitive suite so we could completely avoid the huge OpenSSL dependency...

Erethon, 2023-02-27:
> That's strange. Does LibreSSL not support X25519? Or does it just have a different API? I really wish we had a bare-minimum, well-tested, audited primitive suite so we could completely avoid the [...]

See my [previous reply in this thread](https://github.com/markqvist/Reticulum/issues/240#issuecomment-1444783850).

I don't know if there are other platforms that compile Python against LibreSSL by default (or if the [list in the wikipedia page](https://en.wikipedia.org/wiki/LibreSSL#History) is correct). Given the fact that the 3.7.0 release added support for X25519, I don't think Reticulum needs to do anything about this, the change will eventually end up being picked up by OSes.

Erethon, 2023-02-27:
> Yes, LibreSSL in the latest OpenBSD version doesn't support X25519. See my [previous reply in this thread](https://github.com/markqvist/Reticulum/issues/240#issuecomment-1444783850) for more details.

To back this up, I spun up an OpenBSD host using the latest snapshot (that uses LibreSSL 3.7.1) . Running `rnsd` out of the box failed with the same error about X25519 not being supported. I then upgraded the python `cryptography` package to the `main` branch, since some PRs around [x25519 and libressl](https://github.com/pyca/cryptography/pull/8263) aren't included in a release yet. After the `cryptography` update everything worked, `rnsd` started up and created the keys as usual.

gdt, 2023-02-27:
The peering hash calculation only affects IPv6 link-local addresses.   However, I understand what is going on and just need to chase down what the rules are and where this should be fixed and/or worked around.  It would be interesting to know if it happens on OpenBSD too, but that would be more for extending the workaround to there.   It is good to have confirmation that remote TCP works thought - that means almost everything is ok.

Or you can just turn on debugging and add a print statement.  In peer_announce, add the following two statements just after the line assigning to link_local_address,  but for now comment out the re.sub (which needs import re earlier).
```
            link_local_address = re.sub(r"fe80:[0-9a-f]*::","fe80::", link_local_address)
            RNS.log("peer_announce ifname %s lladdr %s" % (ifname, link_local_address), RNS.LOG_DEBUG)
```
If you see `fe80::` then the bug is not happening and if you see 'fe80:3::` or something (if you are on ifindex 3) then it is.   If you are seeing the `:n::`, then uncommenting the re.sub should work around it.    (I still need to figure out the right fix.)

Erethon, 2023-02-28:
> If you see `fe80::` then the bug is not happening and if you see 'fe80:3::`or something (if you are on ifindex 3) then it is. If you are seeing the`:n::`, then uncommenting the re.sub should work around it. (I still need to figure out the right fix.)

I'm getting `[Debug] peer_announce ifname lo0 lladdr fe80:3::1`, so it's happening in OpenBSD too. This is coming from the `netifaces` package:
```
import netifaces
netifaces.ifaddresses("lo0")
{24: [{'addr': '::1', 'netmask': 'ffff:ffff:ffff:ffff:ffff:ffff:ffff:ffff/128'}, {'addr': 'fe80:3::1', 'netmask': 'ffff:ffff:ffff:ffff::/64'}], 2: [{'addr': '127.0.0.1'}]}
```

Not sure if the fix should be in Reticulum or `netifaces` (which is looking for a new maintainer since 2021).

Erethon, 2023-02-28:
@gdt  you completely [nerd snipped](https://en.wikipedia.org/wiki/Nerd_sniping) me with this one! I dove into a rabbit hole way deeper than I had anticipated with a lot of reading on old documents.

The tl;dr is that you've got the gist of it somewhat right. This is happening because in KAME (and other implementatons maybe?) the `scope id` is embedded into the 3th and the 4th bytes of an address for link local addresses. [Link to FreeBSD handbook](https://docs.freebsd.org/en/books/developers-handbook/ipv6/#ipv6-scope-index) explaining this, [KAME implementation docs](https://github.com/kame/kame/blob/master/IMPLEMENTATION#L475) going in more details, [Wikipedia page](https://en.wikipedia.org/wiki/IPv6_address#Scoped_literal_IPv6_addresses_(with_zone_index)) with a comparison between OSes. My understanding is that this representation (called embedded form) should only by used by the kernel, however some userland programs still use APIs that expose this form to users.

Here's what I think is happening, using as an example a x86 OpenBSD host of mine:
```
thor ~ % ifconfig ale0
ale0: flags=248843<UP,BROADCAST,RUNNING,SIMPLEX,MULTICAST,AUTOCONF6TEMP,AUTOCONF6> mtu 1492
        media: Ethernet autoselect (100baseTX full-duplex)
        status: active
        inet6 fe80::223:54ff:fe19:1f7c%ale0 prefixlen 64 scopeid 0x1
```
Note: The scope id for this is `1`. Now I start `rnsd` and it outputs:
```
[Extra] AutoInterface[Default Interface] Selecting link-local address fe80:1::223:54ff:fe19:1f7c for interface ale0
[Debug] AutoInterface[Default Interface] received peering packet on ale0 from fe80::223:54ff:fe19:1f7c, but authentication hash was incorrect.
```
We can see that the scope id has been inserted in the 3rd and 4th bytes of the address as the spec says! But why hasn't it in the second log line?

The answer is because we're reading data from two different sources. When we're selecting which link-local address to use, we're doing it [via the `netifaces` package](https://github.com/markqvist/Reticulum/blob/master/RNS/Interfaces/AutoInterface.py#L160). On the other hand, when comparing the data with the expected hash we're using the [address returned by the Python socket implementation](https://github.com/markqvist/Reticulum/blob/master/RNS/Interfaces/AutoInterface.py#L248/).

The [socket recvfrom docs](https://docs.python.org/3/library/socket.html#socket-families) specify:
> For [AF_INET6](https://docs.python.org/3/library/socket.html#socket.AF_INET6) address family, a four-tuple (host, port, flowinfo, scope_id)  is used

and here's what the Python socket returns in my host `('fe80::223:54ff:fe19:1f7c', 4917, 0, 1)`. The scope id is returned as the last value of the tuple and not as part of the address.

I haven't 100% verified this, but I believe `netifaces` uses the `getifaddrs` API, which returns the embedded form of the address, hence causing this mess. Fixing `netifaces` would be the cleanest approach, but seeing as there has been no communication from the maintainer in the past two years in issues, maybe a simple workaround like a regex that @gdt proposed earlier is enough.

I wonder if this is an issue on OSX as well and if maybe it's the cause of https://github.com/markqvist/NomadNet/issues/20.

gdt, 2023-02-28:
This is exactly what I meant.   On OpenBSD, does recvfrom(2) return the scopeid as embedded?

What I wasn't sure about is the extend to which the embedded scopeid representation is standardized.  But, getifaddrs(3) on NetBSD returns the LL addr with embedded scope, rather than in scopeid.   netifaces just uses getifaddrs.   And, it just returns an "address" which is sort of a printed representation of `sockaddr_in6.sin6_addr`, and has no concept of being able to return a scopeid.

So I'd say that fixing this is netifaces would require extending it to have a more complicated address representation.  Or perhaps to have it extract scope into sin6_scope and remove it from sin6_addr, thus dropping scope, but that probably has consequences for other users.

More importantly, reticulum is going to have to work with released netifaces until say 6 months after an improved netifaces is released, which is maybe never -- so it needs a workaround.  I'll prepare a PR.

I don't think this is related to the bug you mentioned.   For reasons not clear to me, this problem does not occur on macOS.  netifaces returns a scoped address, but it's `fe80::stuff%en0`, without a numeric embedded scopeid.  netifaces seems to just call getnameinfo, so mac.OS's getnameinfo must use the % form.    Now it's clear: reticulum's code drops the %en0 via split when assigning into adopted_interfaces, and this is why it works on macOS.    So that's where we should also substitute out the embedded scopeid.

gdt, 2023-02-28:
https://github.com/markqvist/Reticulum/pull/246

PR #246 (2023-03-02): AutoInterface: Drop embedded scope identifier on fe80::
The code previously dropped scope identifiers expressed as a trailing "%ifname", which happens on macOS.  On NetBSD and OpenBSD (and likely FreeBSD, not tested), the scope identifier is embedded.  Drop that form of identifier as well, because we keep address and ifname separate, and because the scope identifier must not be part of computing the hash of the address.

Resolves #240, failure to peer on NetBSD and OpenBSD.
commit e273900e (2023-03-02): AutoInterface: Drop embedded scope identifier on fe80::

The code previously dropped scope identifiers expressed as a trailing
"%ifname", which happens on macOS.  On NetBSD and OpenBSD (and likely
FreeBSD, not tested), the scope identifier is embedded.  Drop that
form of identifier as well, because we keep address and ifname
separate, and because the scope identifier must not be part of
computing the hash of the address.

Resolves #240, failure to peer on NetBSD and OpenBSD.
closed 2023-03-02 by markqvist