markqvist/Reticulum #244: request_path not working in examples

faragher, 2023-02-27:
Using the examples, for example echo.py, request_path does not function and hangs indefinitely. Manually announcing when client is open does allow expected communication, but attempting to find a route with only the client and server running does not function.

Which is also confusing as there's no callback for announces in the client, meaning the announces must be silently handled in the background, something unexpected if working from the example code.

Erethon, 2023-02-28:
I don't think that's `request_path` that hangs, although it might look that way. What happens is that the path is requested and the control returns back to while loop and then the client hangs on the `input()` call. 

This patch can better visualize this:
```
diff --git a/Examples/Echo.py b/Examples/Echo.py
index 529dcec..b1499d9 100644
--- a/Examples/Echo.py
+++ b/Examples/Echo.py
@@ -152,6 +152,7 @@ def client(destination_hexhash, configpath, timeout=None):
     # echo request to the destination specified on the
     # command line.
     while True:
+        RNS.log("Entered the while loop!")
         input()
         
         # Let's first check if RNS knows a path to the destination.
@@ -211,6 +212,7 @@ def client(destination_hexhash, configpath, timeout=None):
             # user to wait for an announce to arrive.
             RNS.log("Destination is not yet known. Requesting path...")
             RNS.Transport.request_path(destination_hash)
+            RNS.log("Request path has returned")
 
 # This function is called when our reply destination
 # receives a proof packet.
```

With the output now clearly showing the fact that we're waiting on the `input()` call:
```
python Echo.py 3671bf40132814df485a9ae38744f72b         
[2023-02-28 00:01:34] [Notice] Echo client ready, hit enter to send echo request to 3671bf40132814df485a9ae38744f72b (Ctrl-C to quit)                                                                                                          
[2023-02-28 00:01:34] [Notice] Entered the while loop!                                                                 
                                                                                                                       
[2023-02-28 00:01:35] [Notice] Destination is not yet known. Requesting path...                                                                                                                                                                
[2023-02-28 00:01:35] [Notice] Request path has returned   
[2023-02-28 00:01:35] [Notice] Entered the while loop!                                                                 
```

faragher, 2023-02-28:
Thank you, the patch helped track down my issue. Since the system doesn't provide any feedback on request_path and the echo request is not resent, it simply seems that there's no response from the request. It seems to still cause issues in my program, but if it works in the example and fails in my program, that's my problem, not the bug tracker's.

markqvist, 2023-03-02:
I added a bit of log output to make this less confusing.