markqvist/Reticulum #265: OLED burn-in risk

faragher, 2023-03-28:
Many OLED screens have burn-in issues, and I'm concerned a 24/7 base station will destroy the OLED screen quite quickly. Is it possible to turn off the screen remotely, or set a configuration bit that will turn off the screen after a certain length of time? I realize it's a complicated consideration as to how it turns off and on, but for an always-on unit, it seems only a matter of time before it burns in.

markqvist, 2023-04-02:
There is actually already a command in the firmware for controlling the display. If you send `0xC04500C0`, the display will turn off. Unfortunately, this is not currently persisted in the EEPROM configuration, so it needs to happen at every boot.

Adding a config parameter to the `RNodeInterface` for setting the display intensity would solve the problem easily in the short term. For the next firmware release, we could add a config option to `rnodeconf` for saving the display intensity, and have the firmware persist the config to EEPROM.

pizzadoglover911, 2023-04-02:
Some applications, a time out display that wakes with activity could be useful. Also having the OLED off completely could be helpful as well.

faragher, 2023-04-02:
Thanks, works perfectly. I can iterate over interfaces and shut down by name or RNode class.

markqvist, 2023-05-02:
Just moving this to the RNode repo for ease of workflow.