markqvist/Reticulum #269: "This device contains a valid firmware, but EEPROM is invalid"- After erase of EEPROM

pizzadoglover911, 2023-04-09:
I can assume that the wipe of the EEPROM was successful, but I get an error when trying to re flash the board(as noted in title). The display suggests there is a hardware error after the erasure. 

How can I "correctly initialize the device" as the subsequent error suggests? 

I cannot seem to locate information on the rnodeconf specifics in the manual.

Thanks for the help!

acehoss, 2023-04-09:
The documentation doesn't do a great job of telling you that wiping the EEPROM is a
generally not a good idea. I learned the hard way. You should be able to get your EEPROM reprovisioned using `rnodeconf -a`. This wasn't a total slam dunk for me, but it did rebuild my EEPROM. 

There's a little more info on [this reddit thread](https://reddit.com/r/reticulum/comments/10o9hyu/wiped_eeprom_on_rnode_v1_how_can_i_get_it_going/).

pizzadoglover911, 2023-04-09:
Solved using an alternate machine. Same RNS version, just a x86 vs raspPI. Not sure what was causing the issue at this point.

closed 2023-04-09 by pizzadoglover911