markqvist/Reticulum #270: rnodeconf failing to flash firmware to Lilygo Lora32 v2.1

htd-02, 2023-04-10:
Failing to flash Lilygo Lora32 v2.1 boards with the RNode firmware using the `rnodeconf --autoinstall` command, after some of the flashing is done it ends with
`A fatal error occurred: MD5Sum command returned unexpected result: b'\x91\x18\x99\xa4\xd0\x0e=\xacm2\rr\xff\xfe\xc5'`
The board then always displays a hardware failure when powered on.
Using Debian Bookworm on the Pinebook Pro(arm64), Python v3.10.10(with pyenv), rns v0.5.0.

I did try using an other board of the same model to make sure it wasn't a bricked one and same error occurred.
Also tried flashing from a x64 VM on a desktop, after some of the flashing is done and hanging for a bit it ends with
`A fatal error occurred: Timed out waiting for packet content`

tbeam915, 2023-04-16:
I started a discussion about it here. https://github.com/markqvist/Reticulum/discussions/276

TL;DR
I pulled down a few different versions of the older firmware and tried flashing those with no success.

Swissbandit, 2023-04-18:
i recently got 2 lilygo lora32 v2.1 as t3 v.1.6.1.  flashed without any problem

elena-v2, 2023-04-18:
> If it is a common issue, we may have to add a switch to lower the baudrate for those chips, as you suggest.

I mentioned this in the reticulum matrix room, but it's worth mentioning that this is a well-known issue with many ESP32 devboards, particularly if you have any kind of signal integrity issues. Flashing at 115200 is a pretty safe pick; if it still doesn't work you probably have bigger issues that need addressing.