markqvist/Reticulum #320: Exception in rns service

pizdjuk, 2023-06-03:
after starting following appears:

```
Exception in thread Thread-13:
Traceback (most recent call last):
  File "/usr/lib/python3.8/threading.py", line 932, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.8/threading.py", line 870, in run
    self._target(*self._args, **self._kwargs)
  File "/home/user/.local/lib/python3.8/site-packages/RNS/Transport.py", line 283, in jobloop
    Transport.jobs()
  File "/home/user/.local/lib/python3.8/site-packages/RNS/Transport.py", line 578, in jobs
    packet.send()
  File "/home/user/.local/lib/python3.8/site-packages/RNS/Packet.py", line 261, in send
    if RNS.Transport.outbound(self):
  File "/home/user/.local/lib/python3.8/site-packages/RNS/Transport.py", line 766, in outbound
    tx_time   = (len(packet.raw)*8) / interface.bitrate
```

markqvist, 2023-06-03:
Hmm, I thought I had fixed that some time ago. What version of RNS are you running?

pizdjuk, 2023-06-03:
> Hmm, I thought I had fixed that some time ago. What version of RNS are you running?

0.5.4

pizdjuk, 2023-06-03:
It appears when I play around with system route tables

markqvist, 2023-06-04:
> system route tables

The Reticulum path tables, or IP routes?

pizdjuk, 2023-06-04:
> > system route tables
> 
> The Reticulum path tables, or IP routes?

with iptables on Linux host

markqvist, 2023-06-04:
Moving to discussions then :)