markqvist/Reticulum #342: Nomadnet transfer (page/files) don't work with more than 2 RNodes at the same time

SebastianObi, 2023-07-06:
The error affects several users and can be exactly reproduced.
All details and already performed test scenarios can be found in the discussion.

I hope that Mark will find time to look at this problem.

Thanks

_Originally posted by @SebastianObi in https://github.com/markqvist/Reticulum/discussions/327_

faragher, 2023-07-06:
Related discussion: #327 

Repeatable in high load radio conditions. Appears to be a link timeout issue, but cannot discount buffer issues on the RNode or malformed packets that were stepped on. 

Suggest implementing variable timeout based on baud rate or duty cycle to allow multiple radios to share the frequency in a predictable way.

markqvist, 2023-07-09:
This is not easily solved in Reticulum, since it is a channel utilisation issue. I'll move this issue to the RNode firmware repo, since a better approach is probably to implement a more flexible MA scheme.