markqvist/Reticulum #369: Interoperate public TCP interface

sandrogomes94, 2023-09-13:
I am runing a reticulum instance in an public ngrok TCP interface
and would like interoperate whit the public RNS Testnet Dublin.

instance with my config:

```
[reticulum]
  enable_transport = True
  share_instance = Yes
  shared_instance_port = 37428

[logging]  
  loglevel = 4

[interfaces]
  [[Default Interface]]
    type = AutoInterface
    enabled = No

  [[RNS Testnet Ngrok]]
    type = TCPClientInterface
    enabled = Yes
    target_host = 0.tcp.sa.ngrok.io
    target_port = 19481
```

And another reticulum instance connected to public testnet

instance with testnet config

```
[reticulum]
  enable_transport = True
  share_instance = Yes
  shared_instance_port = 37428

[logging]  
  loglevel = 4

[interfaces]
  [[Default Interface]]
    type = AutoInterface
    enabled = Yes

  [[RNS Testnet Dublin]]
    type = TCPClientInterface
    enabled = Yes
    target_host = dublin.connect.reticulum.network
    target_port = 4965
```

When i try to connect to a destination from one instance to another i get path not found.
I need help with this.

faragher, 2023-09-13:
The two instances have no common connection and will not see each other. (If they're on the same machine this may not be true, but please try the following anyway)

Shared Instances are across the local machine. This means if you open an application, such as Sideband while another, such as NomadNet, is running, then it won't make a new network stack. This is good.

AutoInterface is what connects instances across the local network. As your first config file has this disabled, they won't talk to each other on a local net. This is likely your problem.

If, on the other hand, you're on a single machine, you don't need to run two instances, you can simply have more than one TCPClientInterface. Using these as an example, you can bridge the Dublin node and another network at Ngrok (which is yours? I've never heard of it) with the following:

```
[reticulum]
  enable_transport = True
  share_instance = Yes
  shared_instance_port = 37428

[logging]  
  loglevel = 4

[interfaces]
  [[Default Interface]]
    type = AutoInterface
    enabled = Yes

  [[RNS Testnet Ngrok]]
    type = TCPClientInterface
    enabled = Yes
    target_host = 0.tcp.sa.ngrok.io
    target_port = 19481

  [[RNS Testnet Dublin]]
    type = TCPClientInterface
    enabled = Yes
    target_host = dublin.connect.reticulum.network
    target_port = 4965
```

This will bridge the networks (enable_transport = True) and connect any RNS on your local network (AutoInterface enabled). Now, the AutoInterface is IPv6 stuff I don't understand, but from my end, it works across my network so . . .

Let us know if that works or if I'm misunderstanding your setup.

sandrogomes94, 2023-09-13:
> The two instances have no common connection and will not see each other.

Yes, it make sense

> AutoInterface is what connects instances across the local network. As your first config file has this disabled, they won't talk to each other on a local net. This is likely your problem.

I disabled it because is running in windows.

The principal idea is interoperate diferent tcp interfaces and bridge theys know hosts.

I will test and come back here.

sandrogomes94, 2023-09-13:
Thank you for your reply @faragher 
This worked partially

I test with this setup:

Instance 1
0.tcp.sa.ngrok.io

Instance 2
0.tcp.sa.ngrok.io
dublin.connect.reticulum.network

Instance 3
dublin.connect.reticulum.network

Create a serve destination hash with instance 1 and announce to network

Create a cliente with instance 3 and connect to destination hash of instance 1

I only need wait few minutes to discovery the path

markqvist, 2023-09-13:
Here's another tip. If you want immediate path resolution, take a look at the **Interface Modes** section of the documentation. Transport instances can recursively resolve paths for any clients on their networks:

https://markqvist.github.io/Reticulum/manual/interfaces.html#interface-modes