markqvist/Reticulum #370: Help with rpc request to reticulum instance

sandrogomes94, 2023-09-13:
I am with the idea to create a browser extension to interact with reticulum instance
using the rpc api, but i dont know how pass the auth_key in http request.

faragher, 2023-09-13:
Reticulum doesn't use http requests. If you're going to create an interface, you'll have to create your own program that sends and receives data via Reticulum and converts it to what the browser's expecting. 

I suggest getting used to how to send and receive data, as well as prepare the network stack, by becoming comfortable with these examples: https://github.com/markqvist/Reticulum/tree/master/Examples

The browser side of the program is out of scope for the support in this repository. If you need help interfacing with the RNS side of your program, please let us know in the discussion area.

sandrogomes94, 2023-09-13:
You are right, the multiprocessing module work with socket proccess