markqvist/Reticulum #377: new program "ir"

gdt, 2023-10-01:
When packaging, there is a always a concern with name collisions.  Obviously this is intractable, but "ir" is short and not obviously related to reticulum.  Current files in bin in the package:
```
bin/ir
bin/rncp
bin/rnid
bin/rnodeconf
bin/rnpath
bin/rnprobe
bin/rnsd
bin/rnstatus
bin/rnx
```

I'm not sure what ir does, but perhaps it should be rnir, or something.

markqvist, 2023-10-01:
Well, this is a bit of an oversight on my part. The `ir` program is the Reticulum Identity Resolver.

I was working on its implementation when the recent DDOS attempt against the testnet happened, and then stashed my work on it, to implement control mechanism against such attacks instead. At that point the skeleton code for the utility had already been committed, and I forgot to remove the CLI entry point from the build manifest.

I think the shortness of the command is warranted here, since the intention is also to be able to use it in shell command substitutions like `rnsh $(ir nn-node.home)`. It could be renamed to  `rnir`, which is also pretty short and easy to use, but I have not been able to find any other commands/programs using `ir`, so it doesn't seem like there is any collisions.

Would be interesting with a broader discussion on this. Personally I like the `ir` name, but I am open to doing something else. Maybe we should put this in the discussions for more visibility?

gdt, 2023-10-01:
I think ir only seems reasonable if your head is in Reticulum.    If you are taking the broad view of 20000 packages it is very short and hard to know what it does.   Yes, unix has a 2-letter command tradition, but those are for ls/mv/rm that everybody knows.  I suggested rnir because then all the R commands start with rn, and you find them all at once.

But indeed would be good to see who else thinks what.  (I have no clue about discusions.)

markqvist, 2023-10-01:
Let's just leave it here for now. I'll post a link on Matrix, and anyone interested can weigh in.

faragher, 2023-10-01:
As noted in Matrix: rnir is my vote.

**Argument against "ir":**

Languages, including computer languages, tend to use shorter words for basic, fundamental, or common concepts. Mom or food as opposed to mother-in-law or gasoline. When I see a two letter command, I immediately think of basic commands. Rm, cp, ls, df, or even ip (and may Reticulum grow to that level). Ir isn't that important, even to Reticulum. It's a great command line utility, but probably used less than rnstatus, at least by me. I'd spend my first hundred uses thinking "What command was that again? Lr? Lsr? Ir? Hmm. I'll do ir --help just in case it's actually 'irrevocably destroy" because I've had rm mistakes before." :D

**Case for "rnir":**

Barely longer, and has a _huge_ advantage. Rn isn't exactly a common pair of characters at the start of a word. I can hit "r n tab" and have basically the entire RNS console command list in front of me. It's not a big deal to someone super used to the system, but even at my level, it can be helpful to just hit those three keys and see all my options in front of me.

Which, upon reading the new posts, is the argument made by GDT, so I think we're on the same page.

Erethon, 2023-10-01:
Plus one for `rnir`, it maintains a consistency with the rest of the ecosystem and is easier to package indeed.

markqvist, 2023-10-01:
Thanks for all the input! Lots of great points I hadn't considered. I agree that `rnir` is the best way to go here.

commit f79f1905 (2023-10-01): Changed ir utility name to rnir. Closes #377.
closed 2023-10-01 by markqvist