markqvist/Reticulum #432: RNodeconf cannot reprovision a board.

faragher, 2024-01-18:
RNodeconf refuses to provision a board unless the EEPROM is wiped, which is fair, but also refuses to reprovision if the EEPROM checkum is wrong/missing, which seems incorrect. 

Version 0.7.0

markqvist, 2024-01-18:
Uuuh... That's very silly. Just for information, does running `rnodeconf --eeprom-wipe` manually first allow it reprovision?

markqvist, 2024-01-18:
And could you post what commands you are running specifically, and the output from those?

faragher, 2024-01-18:
```
>rnodeconf -a -U
'clear' is not recognized as an internal or external command,
operable program or batch file.

Hello!

This guide will help you install the RNode firmware on supported
and homebrew devices. Please connect the device you wish to set
up now. Hit enter when it is connected.

Detected serial ports:
  [1] COM8 (None, xxx)

Enter the number of the serial port your device is connected to:
? 1
'clear' is not recognized as an internal or external command,
operable program or batch file.

Ok, using device on COM8 (None, xxx)

Probing device...
[12:12:51] Device connected
[12:12:51] Current firmware version: 1.68
[12:12:51] Trying to read EEPROM...
[12:12:52] EEPROM checksum correct
[12:12:52] Device signature validated

This device is already installed and provisioned. No further action will
be taken. If you wish to completely reinstall this device, you must first
wipe the current EEPROM. See the help for more info.

Exiting now.
```

Note: The "Clear" error is typical, this is running in a Windows environment.

```
>rnodeconf COM8 --eeprom-wipe
[12:15:15] Opening serial port COM8...
[12:15:18] Device connected
[12:15:18] Current firmware version: 1.68
[12:15:18] WARNING: EEPROM is being wiped! Power down device NOW if you do not want this!
```

program successfully clear EEPROM

```
>rnodeconf -a -U
'clear' is not recognized as an internal or external command,
operable program or batch file.

Hello!

This guide will help you install the RNode firmware on supported
and homebrew devices. Please connect the device you wish to set
up now. Hit enter when it is connected.

Detected serial ports:
  [1] COM8 (None, xxx)

Enter the number of the serial port your device is connected to:
? 1
'clear' is not recognized as an internal or external command,
operable program or batch file.

Ok, using device on COM8 (None, xxx)

Probing device...
[12:15:43] Device connected
[12:15:43] Current firmware version: 1.68
[12:15:43] Trying to read EEPROM...
[12:15:44] EEPROM checksum mismatch
```

The -U flag was used in an attempt to force an update. It obviously does not work.

faragher, 2024-02-29:
I'm not sure why this is still open, probably my negligence. I'll fix that, since this was either entirely my error or slightly unclear options. Neither is a reason to keep this open.

closed 2024-02-29 by markqvist

markqvist, 2024-02-29:
Uh, okay, at a second look, that change shouldn't make a difference, and I reverted it. I don't know how the error could happen in the first place, it shouldn't be possible if the EEPROM was erased, and I can't replicate it at all, so I honestly don't know what was up here.

faragher, 2024-03-09:
I did it again. Just ran --eeprom-wipe and now it's telling me the checksum is wrong and failing out. On 0.72. I must just be the enemy of the EEPROM.

faragher, 2024-03-09:
Okay, fascinating new data: I tried wiping the already wiped EEPROM, and everything works again. Perhaps something in the wipe routine?

markqvist, 2024-03-09:
Hmm.. It might be. I've never experienced similar behavior, but I've never actually tried it on Windows, so that may well be the case.

commit 2b764b4a (2024-02-29): Allow EEPROM checksum mismatch on autoinstall. Fixes #432.