markqvist/Reticulum #497: Wrong FW reported in RNodeConf

faragher, 2024-05-04:
**Describe the Bug**

RNodeConf reports incorrect FW version. This also prevents --auto-intstall upgrades

**To Reproduce**

Connect to RNode using RNodeConf (tested on 1.6.1 FW version 1.59). Reports version is, at this time, 1.70 and up to date.

**Expected Behavior**

Reports accurate FW Version

**Logs & Screenshots**

OLED display reports FW 1.59.

Issue on Linux: 

```
rnodeconf 2.1.3
...
[00:47:37] Opening serial port /dev/ttyACM0...
[00:47:40] Device connected
[00:47:40] Current firmware version: 1.70
[00:47:40] Reading EEPROM...
[00:47:41] EEPROM checksum correct
[00:47:41] Device signature validated
```

Windows reports:

```
rnodeconf 2.1.3
...
[18:43:22] Opening serial port com15...
[18:43:26] Device connected
[18:43:26] Current firmware version: 1.59
[18:43:26] Reading EEPROM...
[18:43:26] EEPROM checksum correct
[18:43:26] Device signature validation failed
```

**System Information**
See above

**Additional context**
Issue discovered when tracking down Bluetooth bug. Debugging between the two is overlapping.

markqvist, 2024-05-05:
Ok, pardon the seemingly obvious question here, but I gotta ask, since I've done this myself a few times. Can you confirm that this is not occurring because you actually have another RNode connected to the system, that is using `/dev/ttyACM0`, and that the RNode you are intending to communicate with is on another port? I've done this before, where I completely forgot I had another RNode going off on a long USB cable somewhere, and was seriously scratching my head for a good while.

If that is not the case, something bordering on spooky is going on, because I currently have literally no clue how it would get a firmware version value of `1.70` from the device. It's pulling it via a command from the RNode device, and the value being read in the actual hardware is hardcoded into the compiled firmware.

So if it is indeed an issue with `rnodeconf` or the firmware itself, this is probably going to be an interesting one to hunt down.

faragher, 2024-05-05:
Ahem. Never mind. I won't admit what happened, but it wasn't that I missed an RNode attached. All I can say is I do almost all my work through SSH.

closed 2024-05-05 by faragher

markqvist, 2024-05-05:
Heh, we've all been there :) I once managed to drop an entire database. On the production server - not the local development copy. That required some fast thinking.

faragher, 2024-05-05:
Oof. I've done some DOS `undelete` before, but that's a different level of "Oh no."