markqvist/Reticulum #522: RNCP status error

jacobeva, 2024-07-29:
**Describe the Bug**
rncp shows an incorrect total file size. e.g. transferring a large file using it will show an inaccurate percentage and total size of file (I assume it does not account for the lossless compression in these calculations). I can tell this because it seems to exit the transfer prematurely according to the status, but the file has been successfully copied in its entirety.

**To Reproduce**
Copy any large file (2MB+) using rncp over a link slow enough for you to see the progress.

**Expected Behavior**
The file transfer status should be accurate, showing an accurate total size of the file (maybe the compressed size and then the actual size in brackets) along with an accurate percentage.

**Logs & Screenshots**
Please see the video below:
https://youtu.be/TD4wPWWzXmQ

**System Information**
- Arch Linux - Kernel 6.9.6-arch1-1
- Python 3.12
- rncp 0.7.5

markqvist, 2024-08-17:
Thank you! Will get it fixed :)

markqvist, 2024-08-30:
Fixed in `0.7.6`

closed 2024-08-30 by markqvist

jacobeva, 2024-10-05:
I've just tested this, and it actually still isn't fixed. I'm not sure if it's just the specific file I'm using, but you may want to double check this works properly with a 4MB file (which is what I tested it with just now). It still jumps up to completion. Also, the transfer speed is invalid at the end, it'll say something close to 1mbps (I think it is calculating this assuming the transfer speed was the time from the end of the transmission to it exiting). I can open another issue for the second problem if you want.

markqvist, 2024-10-05:
Ooh, that's some magical RNode you've got there :) Yes, please open a new issue for that, won't forget it then :)

jacobeva, 2024-10-09:
Will do. Could you please re-open this issue as well then? Also, as a pointer, it's not only `rncp` which has this problem, but also Sideband (when sending attachments the percentage sent is incorrect when the file is really big).

markqvist, 2024-10-09:
Ok, I finally found it, no need to create a separate issue. Fix incoming.

commit 8d98c875 (2024-10-09): Fixed resource progress calculation bug. Actually fixes #522.