markqvist/Reticulum #573: Utility issues on Windows an dAndroid

KenSamson, 2024-10-11:
These by themselves are not the end of the world, but left unfixed these accumulate...

Let me know if you need any more info...

**Describe the Bug**

1)  rnsh and rnx on windows are broken  (termios)

2)  rnsh on android returns a wrong version  (0.0.0)

**To Reproduce**

On windows:

Successfully installed rns-0.8.0

C:\Users\ka0pi\.reticulum>rnsh --version
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "C:\Users\ka0pi\AppData\Local\Programs\Python\Python312\Scripts\rnsh.exe\__main__.py", line 4, in <module>
  File "C:\Users\ka0pi\AppData\Local\Programs\Python\Python312\Lib\site-packages\rnsh\rnsh.py", line 37, in <module>
    import termios
ModuleNotFoundError: No module named 'termios'

C:\Users\ka0pi\.reticulum>rnx --version
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "C:\Users\ka0pi\AppData\Local\Programs\Python\Python312\Scripts\rnx.exe\__main__.py", line 4, in <module>
  File "C:\Users\ka0pi\AppData\Local\Programs\Python\Python312\Lib\site-packages\RNS\Utilities\rnx.py", line 31, in <module>
    import tty
  File "C:\Users\ka0pi\AppData\Local\Programs\Python\Python312\Lib\tty.py", line 5, in <module>
    from termios import *
ModuleNotFoundError: No module named 'termios'

Upgraded to 0.8.3
C:\Users\ka0pi\.reticulum>rnsd --version
rnsd 0.8.3

C:\Users\ka0pi\.reticulum>rnsh --version
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "C:\Users\ka0pi\AppData\Local\Programs\Python\Python312\Scripts\rnsh.exe\__main__.py", line 4, in <module>
  File "C:\Users\ka0pi\AppData\Local\Programs\Python\Python312\Lib\site-packages\rnsh\rnsh.py", line 37, in <module>
    import termios

C:\Users\ka0pi\.reticulum>rnx --version
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "C:\Users\ka0pi\AppData\Local\Programs\Python\Python312\Scripts\rnx.exe\__main__.py", line 4, in <module>
  File "C:\Users\ka0pi\AppData\Local\Programs\Python\Python312\Lib\site-packages\RNS\Utilities\rnx.py", line 31, in <module>
    import tty
  File "C:\Users\ka0pi\AppData\Local\Programs\Python\Python312\Lib\tty.py", line 5, in <module>
    from termios import *
ModuleNotFoundError: No module named 'termios'

----------------------

On android, with rns 0.8.3 installed but running against an already running Sideband V 1.1.1:
(Which congains rns 0.8.2)
rnsh --version 

rnsh 0.0.0

rnsd --version

rnsd 0.8.3

**Expected Behavior**
Looks like a set of slight build errors.   It's possible that the fix for android also also takes care of this one.

**Logs & Screenshots**
See above

**System Information**

Windows:
Device name	Coeus
Processor	AMD Ryzen 9 6900HS Creator Edition   3.30 GHz
Installed RAM	32.0 GB (27.7 GB usable)
Device ID	A93BE39E-7E2A-454C-A041-5E88305DE7FF
Product ID	00342-20917-36746-AAOEM
System type	64-bit operating system, x64-based processor
Pen and touch	Touch support with 10 touch points
Edition	Windows 11 Home
Version	23H2
Installed on	‎1/‎30/‎2023
OS build	22631.4169
Serial number	PF3ZEA84
Experience	Windows Feature Experience Pack 1000.22700.1034.0

C:\Users\ka0pi\.reticulum>python --version
Python 3.12.6

Android:   14

Samsung Galaxy

faragher, 2024-10-11:
Problem also exists on Windows 10. I have good news and bad news.

Termios doesn't exist on Windows. Some projects have been started to try to replicate the functions, but they are in various states of usefulness, from the author publicly disavowing Windows and saying nobody should fork their work to rather functional. 

I generally wouldn't recommend trying to port it anyway, since Cygwin exists. It provides a POSIX interface and works as is. 

Unfortunately, Cygwin doesn't support Rust (at least in my experience) so you can't install rnsh on Windows using pip under Cygwin.

So we're kind of stuck. 

I'll look into the termios issue and see if I can think of a way to replace it for Windows, but it's a fundamentally different IO mindset from what I've read. Thanks for bringing it up.

faragher, 2024-10-11:
We might be able to remove the tty dependency from rnx but rnsh is tied to termio and tty pretty heavily. I can't look into it any more tonight, but I'll circle back.

markqvist, 2024-10-11:
Thanks for the details and for bringing up the issue. I've removed the `tty` module dependency from `rnx`, since it was not actually used. It was there to support command-history browsing in interactive mode, but that feature was not completed, and currently disabled anyway. The `rnx` program now runs on Windows, but unfortunately that doesn't make it actually *work*.

Programs such as `rnx` and `rnsh` expect a POSIX interface to things such as pipes, subprocesses and terminal IO, and these things are just fundamentally different on Windows. Making it work would require substantial rewrites, and this is not something that I think makes any sense currently, especially since Cygwin exists, which does provide a full POSIX environment under Windows.

The version display bug on Android `rnsh` should probably be filed over at https://github.com/acehoss/rnsh, since I am not the maintainer of that program.

closed 2024-10-11 by markqvist