markqvist/Reticulum #576: rncp errors

KenSamson, 2024-10-12:
Digest is rejected error, may be a ratchet issue
3 100% transfer reports with 0 bps

All on 0.8.4

Used 40 second to verify there really is not a timeout issue, this fully completes in less than 10 seconds.

Even with the errors this seems to transfer files really well.

~ $ rncp -vvv -w 40 test.txt aeb12b1bbebce4ee60ba6d6641996cd1
[2024-10-12 08:08:17] [Debug] Connected to locally available Reticulum instance via: LocalInterface[37428]
[2024-10-12 08:08:17] [Verbose] Configuration loaded from /data/data/com.termux/files/home/.reticulum/config
[2024-10-12 08:08:17] [Verbose] Loaded 4 known destination from storage
[2024-10-12 08:08:17] [Verbose] Loaded Transport Identity from storage
Path to <aeb12b1bbebce4ee60ba6d6641996cd1> requested ⢄ [2024-10-12 08:08:18] [Debug] Destination <aeb12b1bbebce4ee60ba6d6641996cd1> is now 2 hops away via <80b01d0ce7c1cddcc6a5c6aead778bbf> on LocalInterface[37428]
Establishing link with <aeb12b1bbebce4ee60ba6d6641996cd1>  [2024-10-12 08:08:18] [Error] An error occurred while getting first hop timeout from shared instance: digest sent was rejected
[2024-10-12 08:08:18] [Debug] Link request <372bcc678ea9d50835c587eaa8682d10> sent to <rncp.receive.6f6799caa6dcabc346ab95484e248dd8/aeb12b1bbebce4ee60ba6d6641996cd1>
⡁ [2024-10-12 08:08:18] [Verbose] Link <372bcc678ea9d50835c587eaa8682d10> established with <rncp.receive.6f6799caa6dcabc346ab95484e248dd8/aeb12b1bbebce4ee60ba6d6641996cd1>, RTT is 0.297s
Advertising file resource   [2024-10-12 08:08:18] [Debug] Compressing resource data...
[2024-10-12 08:08:18] [Debug] Compression completed in 0.124 seconds
[2024-10-12 08:08:18] [Debug] Compression saved 474581 bytes, sending compressed
[2024-10-12 08:08:18] [Debug] Starting resource hashmap computation with 4 entries...
[2024-10-12 08:08:18] [Debug] Hashmap computation concluded in 0.001 seconds
[2024-10-12 08:08:18] [Debug] Sent resource advertisement for <5753e3843e560833d245a358b023957596b158461a0d84e6a7790a8e284fde0b>
Transferring file ⢂ 100.0% - 476.20 KB of 476.20 KB - Transferring file ⢁ 100.0% - 476.20 KB of 476.20 KB - Transferring file ⡁ 100.0% - 476.20 KB of 476.20 KB - Transfer complete  100.0% - 476.20 KB of 476.20 KB - 0 bps
test.txt copied to <aeb12b1bbebce4ee60ba6d6641996cd1>
~ $

markqvist, 2024-10-12:
Same as #577, see answer there

KenSamson, 2024-10-12:
Error fixed, now just have progress showing as 100% multiple times and no valid throughput measurements. 

My intent was to use that to help evaluate various rnode settings for real throughput estimates.

As the command works, I understand this is a pretty low priority cosmetic type issue.

~ $ rncp -vvv -w 40 test.txt aeb12b1bbebce4ee60ba6d6641996cd1
[2024-10-12 10:07:57] [Debug] Started shared instance interface: Shared Instance[37428]
[2024-10-12 10:07:57] [Debug] Cleaning ratchets...
[2024-10-12 10:07:57] [Verbose] Bringing up system interfaces...
[2024-10-12 10:07:57] [Debug] Skipping disabled interface "Default Interface"
[2024-10-12 10:07:57] [Debug] Establishing TCP connection for TCPInterface[RNS Testnet Amsterdam/amsterdam.connect.reticulum.network:4965]...
[2024-10-12 10:07:57] [Debug] TCP connection for TCPInterface[RNS Testnet Amsterdam/amsterdam.connect.reticulum.network:4965] established
[2024-10-12 10:07:57] [Debug] Establishing TCP connection for TCPInterface[RNS Testnet BetweenTheBorders/reticulum.betweentheborders.com:4242]...
[2024-10-12 10:07:57] [Debug] TCP connection for TCPInterface[RNS Testnet BetweenTheBorders/reticulum.betweentheborders.com:4242] established
[2024-10-12 10:07:57] [Debug] Establishing TCP connection for TCPInterface[TCP Client Interface/pi.kcmo.net:31313]...
[2024-10-12 10:07:58] [Debug] TCP connection for TCPInterface[TCP Client Interface/pi.kcmo.net:31313] established
[2024-10-12 10:07:58] [Verbose] System interfaces are ready
[2024-10-12 10:07:58] [Verbose] Configuration loaded from /data/data/com.termux/files/home/.reticulum/config
[2024-10-12 10:07:58] [Verbose] Loaded 5 known destination from storage
[2024-10-12 10:07:58] [Verbose] Loaded Transport Identity from storage
Path to <aeb12b1bbebce4ee60ba6d6641996cd1> requested ⢂ [2024-10-12 10:07:59] [Debug] Destination <aeb12b1bbebce4ee60ba6d6641996cd1> is now 2 hops away via <06d09d16b7fdb37ab21e032fba11889b> on TCPInterface[TCP Client Interface/pi.kcmo.net:31313]
Establishing link with <aeb12b1bbebce4ee60ba6d6641996cd1>  [2024-10-12 10:07:59] [Debug] Link request <3455b6d2af09e0aea990c484a084e544> sent to <rncp.receive.6f6799caa6dcabc346ab95484e248dd8/aeb12b1bbebce4ee60ba6d6641996cd1>
⢄ [2024-10-12 10:07:59] [Verbose] Link <3455b6d2af09e0aea990c484a084e544> established with <rncp.receive.6f6799caa6dcabc346ab95484e248dd8/aeb12b1bbebce4ee60ba6d6641996cd1>, RTT is 0.541s
Advertising file resource   [2024-10-12 10:07:59] [Debug] Compressing resource data...
[2024-10-12 10:07:59] [Debug] Compression completed in 0.137 seconds
[2024-10-12 10:07:59] [Debug] Compression saved 474581 bytes, sending compressed
[2024-10-12 10:07:59] [Debug] Starting resource hashmap computation with 4 entries...
[2024-10-12 10:07:59] [Debug] Hashmap computation concluded in 0.002 seconds
[2024-10-12 10:07:59] [Debug] Sent resource advertisement for <8684962c77a4217e836c8f2b0beedb0a0c8fae8b14654edc8e37da83c08b864a>
Transferring file ⡐ 100.0% - 476.20 KB of 476.20 KB - Transferring file ⡠ 100.0% - 476.20 KB of 476.20 KB - Transferring file ⢄ 100.0% - 476.20 KB of 476.20 KB - Transferring file ⢂ 100.0% - 476.20 KB of 476.20 KB - Transfer complete  100.0% - 476.20 KB of 476.20 KB - 0 bps
test.txt copied to <aeb12b1bbebce4ee60ba6d6641996cd1>
[2024-10-12 10:08:00] [Debug] Path request for <9eb0f17e1691e819bb34eddafa2ca82b> on TCPInterface[RNS Testnet BetweenTheBorders/reticulum.betweentheborders.com:4242]
[2024-10-12 10:08:00] [Debug] Ignoring path request for <9eb0f17e1691e819bb34eddafa2ca82b> on TCPInterface[RNS Testnet BetweenTheBorders/reticulum.betweentheborders.com:4242], no path known
[2024-10-12 10:08:01] [Debug] Saved packet hashlist in 1.09ms
[2024-10-12 10:08:01] [Debug] Saving path table to storage...
[2024-10-12 10:08:01] [Debug] Saved 1 path table entries in 6.11ms
[2024-10-12 10:08:01] [Debug] Saving tunnel table to storage...
[2024-10-12 10:08:01] [Debug] Saved 0 tunnel table entries in 1.2ms
[2024-10-12 10:08:01] [Debug] Saving 5 known destinations to storage...
[2024-10-12 10:08:01] [Debug] Saved known destinations to storage in 1.71ms

markqvist, 2024-10-12:
It probably wont be able to get the actual throughput stats either, unless you actually copy over the RPC key config from Sideband. If that still doesn't solve it, please let me know!

markqvist, 2024-10-12:
Or is this without Sideband running? If that is the case, there may be something funky going on on Android in terms of getting the transfer throughput.

closed 2024-10-12 by markqvist