markqvist/Reticulum #577: rnpath on 0.8.4 errors

KenSamson, 2024-10-12:
Android os

Possibly ratchet related?

If you look at the -vvv output of the rncp issue I also filed, this is on that same path.

~ $ rnpath --version                                  rnpath 0.8.4
~ $   rnpath aeb12b1bbebce4ee60ba6d6641996cd1
Path to <aeb12b1bbebce4ee60ba6d6641996cd1> requested ⢁ Traceback (most recent call last):
  File "/data/data/com.termux/files/usr/bin/rnpath", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/data/data/com.termux/files/usr/lib/python3.11/site-packages/RNS/Utilities/rnpath.py", line 513, in main
    program_setup(
  File "/data/data/com.termux/files/usr/lib/python3.11/site-packages/RNS/Utilities/rnpath.py", line 361, in program_setup
    next_hop_bytes = reticulum.get_next_hop(destination_hash)
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/data/data/com.termux/files/usr/lib/python3.11/site-packages/RNS/Reticulum.py", line 1502, in get_next_hop
    rpc_connection = multiprocessing.connection.Client(self.rpc_addr, authkey=self.rpc_key)
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/data/data/com.termux/files/usr/lib/python3.11/multiprocessing/connection.py", line 525, in Client
    answer_challenge(c, authkey)
  File "/data/data/com.termux/files/usr/lib/python3.11/multiprocessing/connection.py", line 776, in answer_challenge
    raise AuthenticationError('digest sent was rejected')
multiprocessing.context.AuthenticationError: digest sent was rejected
~ $

markqvist, 2024-10-12:
I think that's because you have Sideband running (which then provides a shared RNS instance for the system), but you haven't provided the RPC key for this shared instance in the Termux environment. On a "normal" operating system, this would work without any RPC key configuration, but on Android, apps are completely sandboxed, and you need to copy the RPC key from Sidebands "Connectivity" page into `~/.reticulum/config` in the Termux environment.

KenSamson, 2024-10-12:
1000% correct.   All I did was take down sideband, and all is well here.  

Your reply was perfect, I'll try copying that over as well for the education.    

Thank you for all of your hard work!

~ $ rnpath --version                                  rnpath 0.8.4
~ $ rnpath aeb12b1bbebce4ee60ba6d6641996cd1           Path to <aeb12b1bbebce4ee60ba6d6641996cd1> requested ⢂Path found, destination <aeb12b1bbebce4ee60ba6d6641996cd1> is 2 hops away via <06d09d16b7fdb37ab21e032fba11889b> on TCPInterface[TCP Client Interface/pi.kcmo.net:31313]
~ $

markqvist, 2024-10-12:
Also note that when you have Sideband running, and the Termux environment connects via that, it will use the interfaces defined in Sideband of course.

closed 2024-10-12 by markqvist