markqvist/Reticulum #588: Too Many Files Open

faragher, 2024-10-19:
**Describe the Bug**

After ~28 days uptime, Reticulum crashes due to too many files being open, breaking ratchets and logging, and eventually crashing the entire transport instance.

**To Reproduce**

I don't think it wise to wait a month to see if I can duplicate, this is a fundamental issue.

**Expected Behavior**

File handles closed after use.

**Logs & Screenshots**

Typical: 

```
Oct 18 05:52:07 reticulum.betweentheborders.com rnsd[553]:   File "/home/xxx/.local/lib/python3.11/site-packages/RNS/Identity.py", line 288, in persist_job
Oct 18 05:52:07 reticulum.betweentheborders.com rnsd[553]: OSError: [Errno 24] Too many open files: '/home/xxx/.reticulum/storage/ratchets/xxx>
```

**System Information**
- Debian 
- Python 3.11.2
- RNSD 0.7.7

**Additional context**

Working theory is that there's an error inside a try/except cycle which leaves a handle dangling.

Suggestion: Either migrate to `with open('file.ext','r') as f:` format or move file.close() out of `try:` and into `finally:`

faragher, 2024-10-19:
Okay, I found the issue. Identity.py line 332 opens a file which is never closed. I'm halfway through changing Identity.py over into `with open` so I'm going to finish that up and fix the problem. If that's unacceptable, then it's a quick fix closing the file that was opened on 332.

liamcottle, 2024-10-19:
> and eventually crashing the entire transport instance.

Heh, I've noticed now and then when I get up in the morning, MeshChat is no longer running and had closed itself overnight. I hadn't added any logging for this yet, but this _could_ be related. Seemed to only happen once a month or so... :D

Looks like your error log is related to linux, whereas I'm running on Windows. So no idea until I investigate further, but thought I'd throw in some extra info...

closed 2024-10-20 by markqvist

markqvist, 2024-10-20:
Yep, that was a very classical oversight of forgetting to close the file handle again, and even in such an often-used function. Oof! Thanks for the report and fixing it. I'll release an update for this shortly! I'm betting it's the same cause for your crash @liamcottle.

faragher, 2024-10-20:
@liamcottle  I would assume (a dangerous word) Windows has a similar file handle limit, which may or may not be even the same magnitude as Linux, but I'm willing to bet it would be a problem eventually. Given the similar timing, it wouldn't surprise me, but going back a few versions, sometimes things happened without any error logs I've noticed.

That being said, I can now target these crashes pretty well: I noticed it was down when the Amber Pages said everything was "two days ago." Gave me a ~24 hour period to look at in the logs.

So while this was a problem that has been resolved, I think I have an edge on future troubleshooting, as I think that system is on debug logging. I'll check when I switch over to the new build.

PR #589 (2024-10-20): Fixed file access bug, added fail-safe access
Should fix #588 I did some testing, but I don't think I managed to enable ratchets, so I'm unsure it's entirely solid. It's a simple drop-in replacement so there should be little problem, but I want to be upfront about it.

The change around line 334 is the only one actually required for the bug fix, but I consider the preventative fixes worthwhile. Let me know if you'd like to to go through the entire project.