markqvist/Reticulum #607: Latest `cryptography` package on Python doesn't install on 32bit armv7

deavmi, 2024-11-10:
**Read the Contribution Guidelines**
Before creating a bug report on this issue tracker, you **must** read the [Contribution Guidelines](https://github.com/markqvist/Reticulum/blob/master/Contributing.md). Issues that do not follow the contribution guidelines **will be deleted without comment**.

- The issue tracker is used by developers of this project. **Do not use it to ask general questions, or for support requests**.
- Ideas and feature requests can be made on the [Discussions](https://github.com/markqvist/Reticulum/discussions). **Only** feature requests accepted by maintainers and developers are tracked and included on the issue tracker. **Do not post feature requests here**.
- After reading the [Contribution Guidelines](https://github.com/markqvist/Reticulum/blob/master/Contributing.md), delete this section from your bug report.

**Describe the Bug**
A clear and concise description of what the bug is.

**To Reproduce**
Get an `armv7` machine, so a 32-bit arm chip. This is possible with hardware (like mine) usign the Raspberry Pi 2.

**Expected Behavior**
`pip3 install rns` should install easily.

**Logs & Screenshots**
If you try and install then you will get a build error. The minimum `cryptography` (pip package) requirement is `>=` but for some reason the *much later* python pip package for probably non-64bit platforms requires building a Python wheel (whatever it is called) locally. This requires `libffi-dev` to be installed and then Rust.

I got those all done but then:

[image]

**System Information**
- OS and version: Ubuntu Server 32 bit
- Python version: 3
- Program version: latest

**Additional context**
Add any other context about the problem here.

deavmi, 2024-11-10:
I have been working on a Docker image in general and for the quirks needed for this platform

deavmi, 2024-11-10:
Getting `maturin` built...

Will keep updating this thread and make a PR if necessary

deavmi, 2024-11-10:
Looks like it won't get passed the maturin stuff. Idk.

The `cryptography` site doesn't help much. Talk about pushing rust everywhere and making installations impossible :roll_eyes:

deavmi, 2024-11-10:
Going to try some other things maybe... one thing could be choosing (by my choice) a version of the `cryptography` package that means `rns`'s requirements but is older than the current latest version which uses Rust.

I did end up ordering a new 64bit based Raspberry Pi in the meantime

Erethon, 2024-11-10:
I think this should be better tracked in the cryptography repo, as it doesn't have much to do with Reticulum. Having said that, you can use the `rnspure` package instead that doesn't depend on cryptography. The reticulum readme has more information on `rnspure`.

deavmi, 2024-11-10:
Oh! Thanks. I will take a look.

And yes you are correct, this is indeed a misplaced complaint of mine.

----
Tristan B. Velloza Kildaire

Web: https://deavmi.assigned.network
BNET IRC: deavmi
Email: ***@***.***

deavmi, 2024-11-13:
The `rnspure` route seems to work. Very thankful @Erethon

closed 2024-11-13 by deavmi

markqvist, 2024-11-15:
As [detailed in the manual](https://reticulum.network/manual/gettingstartedfast.html#raspberry-pi), it's recommended to run RNS on 64-bit Raspberry Pi systems, exactly because of this annoying issue.

On 32-bit machines, you can try to use the distro-provided `PyCa/cryptography` package, by installing that prior to RNS installation with something like `sudo apt install python3-cryptography`. Whether that works or not has been a bit of a rollercoaster over the past years (hence the recommendation to just use 64-bit), and if it doesn't, the only route is getting it to compile manually, which can be pretty cumbersome, due to frequent changes in the build system of the `cryptography` package.

deavmi, 2024-11-16:
>due to frequent changes in the build system of the cryptography package.

Indeed! It was complicated.

----
Tristan B. Velloza Kildaire

Web: https://deavmi.assigned.network
BNET IRC: deavmi
Email: ***@***.***