markqvist/Reticulum #623: 'rnsd' requires source files to properly import

gretel, 2024-11-25:
**Describe the Bug**
Stepped on this while working on `openwrt` package for `rns`.
By default, `openwrt` [does package bytecode only](https://github.com/openwrt/packages/tree/master/lang/python#build-considerations) (`.pyc`) files.

**To Reproduce**
It's bit hard to reproduce right now. But basically deleting all `.py` files in the `packages` path should do the trick, i.e.:

```shell
cd /usr/lib/python3.11/site-packages/RNS/
find . -name '*.py' -type f -delete
```

**Expected Behavior**
`rnsd` should work even when the source files are not present.

**Logs & Screenshots**
```shell
root@ap-in:~# opkg install rns
Installing rns (0.8.6-r2) to root...
Downloading http://bld.jitter.local:8080/openwrt/bin/targets/x86/64/packages//rns_0.8.6-r2_x86_64.ipk
Configuring rns.
Collected errors:
 * resolve_conffiles: Existing conffile /etc/config/rnsd is different from the conffile in the new package. The new conffile will be placed at /etc/config/rnsd-opkg.
root@ap-in:~# rnsd -vvv
[2024-11-25 02:59:08] [Error] Local shared instance appears to be running, but it could not be connected
[2024-11-25 02:59:08] [Error] The contained exception was: name 'LocalInterface' is not defined
[2024-11-25 02:59:08] [Verbose] Bringing up system interfaces...
Traceback (most recent call last):
  File "/usr/bin/rnsd", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/usr/lib/python3.11/site-packages/RNS/Utilities/rnsd.py", line 71, in main
  File "/usr/lib/python3.11/site-packages/RNS/Utilities/rnsd.py", line 41, in program_setup
  File "/usr/lib/python3.11/site-packages/RNS/Reticulum.py", line 285, in __init__
  File "/usr/lib/python3.11/site-packages/RNS/Reticulum.py", line 466, in __apply_config
NameError: name 'Interface' is not defined. Did you mean: 'interface'?
root@ap-in:~# opkg install rns-src
Installing rns-src (0.8.6-r2) to root...
Downloading http://bld.jitter.local:8080/openwrt/bin/targets/x86/64/packages//rns-src_0.8.6-r2_x86_64.ipk
Configuring rns-src.
root@ap-in:~# rnsd -vvv
[2024-11-25 02:59:18] [Debug] Started shared instance interface: Shared Instance[37428]
[2024-11-25 02:59:18] [Debug] Cleaning ratchets...
[2024-11-25 02:59:18] [Extra] Cleaning resource and packet caches...
[2024-11-25 02:59:18] [Verbose] Bringing up system interfaces...
[2024-11-25 02:59:19] [Extra] AutoInterface[Default Interface] No link-local IPv6 address configured for lo, skipping interface
[2024-11-25 02:59:19] [Extra] AutoInterface[Default Interface] Selecting link-local address fe80::20d:b9ff:fe43:578d for interface br-lan
[2024-11-25 02:59:19] [Extra] AutoInterface[Default Interface] Creating multicast discovery listener on br-lan with address ff12:0:d70b:fb1c:16e4:5e39:485e:31e1
[2024-11-25 02:59:19] [Verbose] AutoInterface[Default Interface] discovering peers for 1.5 seconds...
[2024-11-25 02:59:20] [Debug] Skipping disabled interface "UDP Interface"
[2024-11-25 02:59:20] [Debug] Skipping disabled interface "TCP Server Interface"
[2024-11-25 02:59:20] [Debug] Skipping disabled interface "TCP Client Interface"
[2024-11-25 02:59:20] [Debug] Skipping disabled interface "I2P"
[2024-11-25 02:59:20] [Debug] Skipping disabled interface "RNode LoRa Interface"
[2024-11-25 02:59:20] [Debug] Skipping disabled interface "Packet Radio KISS Interface"
[2024-11-25 02:59:20] [Debug] Skipping disabled interface "Packet Radio AX.25 KISS Interface"
[2024-11-25 02:59:20] [Verbose] System interfaces are ready
[2024-11-25 02:59:20] [Verbose] Configuration loaded from /root/.reticulum/config
[2024-11-25 02:59:20] [Verbose] Loaded 0 known destination from storage
[2024-11-25 02:59:20] [Verbose] Loaded Transport Identity from storage
[2024-11-25 02:59:20] [Notice] Started rnsd version 0.8.6
```

**System Information**
- OpenWRT SNAPSHOT
- Python 3.11.10
- Reticulum 0.8.6

markqvist, 2024-11-25:
This is probably due to the local import of the `Interface` class in all interface modules:

```python
from .Interface import Interface
```

I'm pretty sure there *was* a reason I did this originally, and I vaguely remember it being something about Android compatibility, but it honestly evades me right now.

Just to test it, could you try changing it to the below, maybe just in `AutoInterface.py` and `LocalInterface.py`?

```python
from RNS.Interfaces.Interface import Interface
```

If that makes it work, I'll change it to that and figure out any potential platform issues on other platforms with conditionals of some kind. It's better to import it directly from the package namespace anyway.

liamcottle, 2024-11-27:
> If that makes it work, I'll change it to that and figure out any potential platform issues on other platforms with conditionals of some kind.

Please be careful with conditional imports 🙏  This was a major issue when getting MeshChat to build with cxfreeze.

https://github.com/liamcottle/reticulum-meshchat/blob/master/setup.py#L22
https://github.com/marcelotduarte/cx_Freeze/discussions/2039
https://github.com/marcelotduarte/cx_Freeze/issues/2041

markqvist, 2024-11-27:
@gretel, what is your take on the OpenWRT packaging? Is it even worth spending time on an `opkg` distribution if they are moving to `apk` now anyways? Or will it still be valuable for the many existing OpenWRT installs based on `opkg`?

gretel, 2024-11-29:
> @gretel, what is your take on the OpenWRT packaging? Is it even worth spending time on an `opkg` distribution if they are moving to `apk` now anyways? Or will it still be valuable for the many existing OpenWRT installs based on `opkg`?

my take is... [not ready](https://github.com/gretel/feed-reticulum) 😄 

luckily, there is no change required to build for the `alpine package manager`. actually it depends on the `openwrt` version.

to support existing (`stable`) versions of `openwrt` building for the existing targets is required. systems built recently do require alpineish packages.

big issue is the cross compilation part. this requires quite some resources. i'm looking into a `ci/c` solution right now.

btw, what's your platform?

gretel, 2024-11-30:
> Just to test it, could you try changing it to the below, maybe just in `AutoInterface.py` and `LocalInterface.py`?
>
> ```python
> from RNS.Interfaces.Interface import Interface
> ```
>
> If that makes it [...]

now that i intend to do i figured i can not because there is no source code to change 😆
this issue is about only compiled bytecode being present, no source, like in `openwrt` packages.

markqvist, 2024-11-30:
> btw, what's your platform?

armv7l (64-bit Qualcomm IPQ6000)

I've changed the interface imports in the latest commits, so it should work now. No conditionals needed either, so this will just be the standard way going forward.

Can you compile it from the latest source tree here in the repo, or do you need a `.whl` to do it? I package one if so.

gretel, 2024-11-30:
> armv7l (64-bit Qualcomm IPQ6000)
>
> I've changed the interface imports in the latest commits, so it should work now. No conditionals needed either, so this will just be the standard way going [...]

thanks @markqvist - the toolchain i brought up makes it easy now to reproduce. no, nothing required, the [package](https://github.com/gretel/feed-reticulum) is made from the `pip`.

will try and report back.

markqvist, 2024-11-30:
Awesome! Thanks a lot for working on this! Having a proper OpenWRT package will be very nice.

gretel, 2024-11-30:
did not understand why you were offering `.whl`. Think i [get it](https://github.com/markqvist/Reticulum/pull/626) now 😆

gretel, 2024-11-30:
built a fictious [0.8.7-beta](https://github.com/gretel/Reticulum/releases/tag/0.8.7-beta) but to my disappointment it renders same result 😢 

```
root@ap-in:~# apk add rns
fetch http://bld.jitter.local:8080/openwrt/bin/packages/x86_64/routing/packages.adb
(1/2) Installing python3-cryptography (41.0.7-r1)
Executing python3-cryptography-41.0.7-r1.post-install
(2/2) Installing rns (0.8.7-r1)
rns-0.8.7-r1: installing file to etc/config/rnsd
Executing rns-0.8.7-r1.post-install
OK: 102 MiB in 297 packages
root@ap-in:~# rnsd --version
rnsd 0.8.7
root@ap-in:~# rnsd -vvv
[2024-12-01 00:41:49] [Error] Local shared instance appears to be running, but it could not be connected
[2024-12-01 00:41:49] [Error] The contained exception was: name 'LocalInterface' is not defined
[2024-12-01 00:41:49] [Verbose] Bringing up system interfaces...
Traceback (most recent call last):
  File "/usr/bin/rnsd", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/usr/lib/python3.11/site-packages/RNS/Utilities/rnsd.py", line 71, in main
  File "/usr/lib/python3.11/site-packages/RNS/Utilities/rnsd.py", line 41, in program_setup
  File "/usr/lib/python3.11/site-packages/RNS/Reticulum.py", line 285, in __init__
  File "/usr/lib/python3.11/site-packages/RNS/Reticulum.py", line 466, in __apply_config
NameError: name 'Interface' is not defined. Did you mean: 'interface'?
```

markqvist, 2024-12-01:
Ok, I missed that one. It's fixed now in 19adadf4cfa8e3df44507b472d1732cf00d5cae2.

But I'm worried there's others it might choke on, I'm finding it a bit hard to predict which relative imports will work, and which won't, since apparently `from .vendor.platformutils import get_platform` is fine, whereas `from .Interfaces.Interface import Interface` is not...

Either way, you're welcome to give it a try again now, hopefully that will be all that is needed. I'm trying to get a build environment set up for OpenWRT locally, but not sure yet if I succeeded. It's churning... Any way to only build packages, and not build an entire firmware image every time?

gretel, 2024-12-01:
> Either way, you're welcome to give it a try again now, hopefully that will be all that is needed.

built another package and tried, but no luck. got bit paranoid and manually ensured the bytecode is recent.

```
root@ap-in:/usr/lib/python3.11/site-packages/RNS# apk add rns
fetch http://bld.jitter.local:8080/openwrt/bin/packages/x86_64/routing/packages.adb
WARNING: updating and opening http://bld.jitter.local:8080/openwrt/bin/packages/x86_64/routing/packages.adb: remote server returned error (try 'apk update')
fetch http://bld.jitter.local:8080/openwrt/bin/packages/x86_64/telephony/packages.adb
WARNING: updating and opening http://bld.jitter.local:8080/openwrt/bin/packages/x86_64/telephony/packages.adb: remote server returned error (try 'apk update')
(1/2) Installing python3-cryptography (41.0.7-r1)
Executing python3-cryptography-41.0.7-r1.post-install
(2/2) Installing rns (0.8.7-r2)
Executing rns-0.8.7-r2.post-install
OK: 102 MiB in 297 packages
root@ap-in:/usr/lib/python3.11/site-packages/RNS# rnsd -vv
[2024-12-01 20:47:57] [Error] Local shared instance appears to be running, but it could not be connected
[2024-12-01 20:47:57] [Error] The contained exception was: name 'LocalInterface' is not defined
[2024-12-01 20:47:57] [Verbose] Bringing up system interfaces...
Traceback (most recent call last):
  File "/usr/bin/rnsd", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/usr/lib/python3.11/site-packages/RNS/Utilities/rnsd.py", line 71, in main
  File "/usr/lib/python3.11/site-packages/RNS/Utilities/rnsd.py", line 41, in program_setup
  File "/usr/lib/python3.11/site-packages/RNS/Reticulum.py", line 285, in __init__
  File "/usr/lib/python3.11/site-packages/RNS/Reticulum.py", line 466, in __apply_config
NameError: name 'Interface' is not defined. Did you mean: 'interface'?
```

> I'm trying to get a build environment set up for OpenWRT locally, but not sure yet if I succeeded. It's churning...

it's hard in the beginning.. lot's of `Makefile` craze.

> Any way to only build packages, and not build an entire firmware image every time?

yes, that's possible, i.e.:

```shell
./scripts/feeds update reticulum
./scripts/feeds install -p reticulum -a

make package/feeds/reticulum/rns/clean
make package/feeds/reticulum/rns/compile NO_DEPS=1
```

maybe we can reproduce the issue locally like by deleting the `.pyc` files?

markqvist, 2024-12-02:
> found this: https://pypi.org/project/pyc-wheel/

Nice, that was a great help. Was able to reproduce the issue locally and correct. The underlying issue was simply that the dynamic sub-module loader for the package was not taking `.pyc` files into account when creating the package structure, thus making some sub-modules unimportable.

I'm testing this now across all platforms, and will push shortly.

gretel, 2024-12-02:
> Nice, that was a great help. Was able to reproduce the issue locally and correct. The underlying issue was simply that the dynamic sub-module loader for the package was not taking `.pyc` files into [...]

that's great news!

markqvist, 2024-12-02:
Ok, it's pushed now in 8df89cc2d0a396bd8127ec3ab13e98df42dc37f2. Haven't yet tried compiling the OWRT package, but I'm betting it'll work now :)

gretel, 2024-12-02:
you need to add the feeds having the packages and this one https://github.com/gretel/feed-reticulum - but sure, i'm going to proceed as my time allows.

gretel, 2024-12-02:
confirming fix 🥳 well done!

```bash
root@ap-in:~# apk update
OK: 684 distinct packages available
root@ap-in:~# apk add rns
(1/2) Installing python3-cryptography (41.0.7-r1)
Executing python3-cryptography-41.0.7-r1.post-install
(2/2) Installing rns (0.8.7-r3)
Executing rns-0.8.7-r3.post-install
OK: 102 MiB in 297 packages
root@ap-in:~# pgrep rnsd
30602
```

closed 2024-12-02 by gretel

markqvist, 2024-12-02:
Nice work! Great to see. Could you share the package on a pre-release on https://github.com/gretel/feed-reticulum or something like that? Would like to try it out.

gretel, 2024-12-03:
https://github.com/marketplace/actions/openwrt-packager 🤷🏼

gretel, 2024-12-04:
https://github.com/gretel/reticulum-openwrt 🤷🏼