markqvist/Reticulum #717: Announcements stored for enough time can be rebroadcasted to poison routing

regulad, 2025-02-16:
## Issue:  

1. In the current implementation, only the last 64 announcement-specific blobs are kept in memory to see if an announcement has previously been seen before. This means that if you listen to the network for the time it takes for 65 announcements to be emitted, you can start emitting cryptographically valid announcements from long enough ago that nodes will no longer drop them as already processed.

https://github.com/markqvist/Reticulum/blob/master/RNS/Transport.py#L82
https://github.com/markqvist/Reticulum/blob/master/RNS/Transport.py#L1608
https://github.com/markqvist/Reticulum/blob/master/RNS/Transport.py#L1487

2. Therefore, any adversarial node with enough memory to capture 65+ announces can poison the network's routing and stop packets from reaching their destination efficiently.

https://matrix.to/#/@qubic:qubicfluff.pl on Matrix identified this issue, and shared a reproducable proof-of-concept adversarial node acting on the exploit written in Rust: https://matrix.to/#/!TRaVWNnQhAbvuiSnEK:matrix.org/$6QRo1RacbI4D5doAp9yOWaN6R7GQPWGxgHE2kmYsZxM?via=matrix.org&via=tchncs.de&via=tapenet.org  

The importance of this vulnerability largely depends on the extent to which an adversarial node fits into Renticiulum’s threat model. However, it’s also possible that in a sufficiently large network, there could be a delay in the propagation of announcements, allowing for 64 unrelated announcements to be generated and transmitted into the mesh before an announcement reaches a node that had already seen it, provided the announcement had gone through less than m hops. 

## Proposed solution:  

#### Please note that this is NOT the only way this problem can be solved, and is simply just MY PERSONAL proposal after chatting in the chatroom with some other RNS users on Matrix. I am aware that feature requests don't belong in the issues according to CONTRIBUTING.md, but I believe it is not a problem considering it is in response to an issue I substantiated above.

1. Transport nodes need a way to verify that an announcement was minted recently. If they know an announcement was minted in the recent past, then replayed announcements can be dropped. This leaves the 64-past-announcement system with a single job, stopping announcements from traveling in a loop between nodes, its original purpose. To implement this, add a 64-bit integer to the announcement body and ensure it is included when signing the message. Routers can then check that the timestamp meets the following criteria:

    a) The timestamp in the packet is not ahead of the router's clock.

    b) (announcement timestamp) - (clock timestamp) < (hops) * (expected time per hop), where expected time per hop can be configured by the router. Fixed high-bandwidth routers can be more forgiving in their expected time per hop, and lower-bandwidth connections like LoRa & Packet radio can be more stringent with the timing of the hop.

2. This implementation protects against announcement replays, but it makes a dangerous assumption that all nodes in the network share the same time. An assumption like this is impossible in a system without any centralized authority that can serve something like an NTP server. 
 
3. Therefore, time will need to be synchronized through the mesh. This proposal does not provide the details of any mesh time-sharing implementation, as any suitable time-sharing protocol will be exceedingly complex and need to account for many factors, principally network solubility, high ping, and completely unpredictable jitter that may exist between any two routers. 

4. To produce a clock suitable for verifying the novelty of an announcement, the following three assumptions will need to be made by any node participating in time sharing. It is possible that future implementation of time sharing will not make these assumptions or will only make a subset of them, but for the explicit purpose of this proposal, they will be used.

    a) **A second lasts ~1 second**. Unfortunately, this means that nodes traveling at relativistic speeds may need to sit out for now. The reason for making this assumption is simple: if an adversary could host enough nodes to adjust the perceived rate at which time moves, they could widen the window for a replay by slowing down the advancement of time. (alternatively, they could speed up time and cause the rollover to happen sooner) Systems with clocks susceptible to slipping on perceptible timescales may join the network, so perhaps tolerance for fast/slow clocks could be included. A system with a highly inaccurate clock may need to “play the table” and sync with the network frequently to continue issuing announcements that the network can perceive as valid.

    b) **Time only moves forward.** This assumption must be made for the same reason as assumption a, as any adversary could declare that the current time is a time in the past and replay an announcement. With a 64-bit integer that tracks time, the rollover time will be 2^64 seconds. It would, therefore, be impractical for any adversarial node to hold onto an announcement for 2^64 seconds to mount a replay attack. Because of the previously mentioned challenges in implementing a time-sharing protocol, making sure all nodes reach a consensus on a certain time before the clock starts advancing may be difficult.

    c) **Mesh time can only be used to measure elapsed time and has no relation to the actual timestamp.** This assumption implicitly protects the network clock from a 51% attack, as it makes no difference who sets the epoch. So long as time marches forward at the same pace, and nobody, not even a collective, can change this fact, the clock can still be used to verify the elapsed time.  

### Conclusion:  

There is a much simpler solution to setting a clock that can verify the novelty of announcements: ask the user to set the exchange clock times out-of-band and then trust the system clock. Whether implementing a network time-sharing protocol is worth the effort is beyond me; the purpose of this issue is to bring attention to the fact that some sort of network-synchronized time is necessary to prove the novelty of announcements and, therefore, protect the network.

faragher, 2025-02-16:
This shouldn't be possible. (I'm not saying you didn't do it, I'm saying there may be an implementation error)

Announcements contain a timestamp (or serial number for devices without RTCs) and thus shouldn't be susceptible to replay attacks. If the system only checks the last n announcements for uniqueness, then this is absolutely a problem. However, for well-formed announcements it should be sufficient to check that the time bytes are higher than any known announce (as well as checking for uniqueness). 

Local time from the announcer is all that's required, as the sequence is all that's important, not the actual time of announce.

This sequence is found in the second set of five bytes in the "random" field on the wire:https://github.com/markqvist/Reticulum/wiki/Wire-Packet-Formats

If changing the system to also check the announces are in the proper sequence doesn't solve the issue, then we can examine more complicated solutions, but this attack is likely mitigated by a minor logic change. Good catch!

regulad, 2025-02-16:
I see. I was unaware that nodes tracked the previous local times of receiving announcements. Still, I think there may be a future purpose for a network clock as the number of nodes and, therefore, memory required goes up. Would be good to see if anybody can reproduce the bug using another PoC.

AwesomeQubic, 2025-02-16:
It only checks for last 64 blobs in announcements

faragher, 2025-02-16:
> It only checks for last 64 blobs in announcements

Fantastic, that will have to be addressed, thank you.

> I see. I was unaware that nodes tracked the previous local times of receiving announcements. Still, I think there may be a future purpose for a network clock as the number of nodes and, therefore, memory required goes up. Would be good to see if anybody can reproduce the bug using another PoC.

I agree network time is very helpful. However NTP is application level, not stack level. It also requires trust and fairly prompt messages, which Reticulum cannot promise.

Consider the following use cases: on a network-enabled PC or mobile device, rather accurate network time can be gathered from multiple sources, including Internet NTP and cell towers. 

On a sandboxed network, a central NTP of arbitrary accuracy is sufficient, and if that needs to be linked to global network time, it's simple to set up a stratum 1 GPS time system with between 10 USD (for the GPS system) to $75 USD (for the GPS, a SBC, case, and power supply).

On a board with a GPS (such as the T-Beam) a system with sub-second time precision can be created without even using the PPS pin. (which is apparently just for a flashing light, thanks LilyGO)

Any system with an RTC only needs to be calibrated once per power cycle, assuming required precision is in seconds and not milliseconds.

So a non-networked board with no GPS capability and either no RTC or for initial RTC sync is the central use case we'd be looking at. Since you're communicating with another system, one would assume there's a LoRa or other real-time, non-typical connection. Setting an application to beacon a time over a Plain destination would take care of these systems without any more trust than you're already showing a radio station, and it wouldn't propagate over the wider network.

Of course if you're looking for a truly sandboxed system ingesting only paper messages, then there are wider considerations regarding trusted communications, and the above systems to sync network time still exist.

If you're at all interested in network time and aren't familiar with stratum 1 GPS time systems, it's easy to set up and a really useful exercise. I use an internal NTP that provides me with the same or better accuracy I get from places like NIST, and it cost me $10. It's fun.

regulad, 2025-02-16:
It would need to be some sort of consensus-based time synchronization,
possibly based on the timestamp in announces as this would minimize
overhead. No reason it has to be an app and not a guarantee of the network,
especially if it’s used to verify network packets. Maybe make a discussion
for it? I don’t want this issue to get clogged talking about time syncing
nor do I want talking about time syncing to get lost when this issue is
closed!

LinuxinaBit, 2025-02-16:
The 64 blob limit on the comparison mechanism makes sense.
The biggest problem is that the timestamp portion of the blob isn't checked against the previous one to make sure the announce is actually newer.
Consensus-based time synchronization is an interesting idea, but I don't think it's necessary. The remaining issues once the timestamp is actually verified to be newer than the previous one are minimal and (I don't think) would be solved using time synchronization.

regulad, 2025-02-16:
Just a quick question, wouldn’t you still have a problem with the size of
the local times of remote notes in memory growing linearly with the number
of nodes? Tracking blobs should only be for announce propagation loops, and
a per-node time validation seems like a short-term-only solution.

LinuxinaBit, 2025-02-16:
> Just a quick question, wouldn’t you still have a problem with the size of the local times of remote notes in memory growing linearly with the number of nodes?

@regulad The 'local time' is just the timestamp of the previous announce.
You only need to store the most recent timestamp in order to verify new ones, and it's already contained within the 'random blob' as faragher mentioned.
All we need to do is _actually verify_ that the blob contained in incoming announces is actually newer than the most recent blob.
This almost entirely fixes this issue, as I mentioned here: https://github.com/markqvist/Reticulum/issues/717#issuecomment-2661644404

markqvist, 2025-02-17:
Thanks for the detailed issue, proposed solutions, and input and comments from everyone. Reading it all now.

markqvist, 2025-02-17:
As @faragher says:

> Local time from the announcer is all that's required, as the sequence is all that's important, not the actual time of announce.

This is correct, and we need to avoid the requirement of a full network-wide timebase consensus, as that would make it very slow and difficult for nodes not equipped with an RTC to join the network.

This is why the "random blob" design simply focused on having *some sort* of timebase for the announces, and only having the requirement that they *increment* on each successive announce. On a system with an RTC, the easiest approach is simply to use the RTC seconds for this. On a system without an RTC, a simple counter can be used, provided this value can be stored between reboots of the node.

> The main cause of the problem is that the timestamp portion of the blob isn't checked against the previous timestamp to make sure the new announce is actually newer.

I agree with @LinuxinaBit here. Simply adding `timebase` field to the path table entries for destinations will solve the issue, by letting Transport check whether a received announce lies too far in the "past" - whatever that means to the node that emitted it, and whether it is based on actual time, or simply an incrementing counter.

So far so good. Can anyone come up with faults in, or viable attack vectors against that approach?

---

The above is the most important part, but here's some additional considerations.

Now, this approach *does* introduce another inconvenience, and that is that an announce which was inadvertently emitted with a "future blob", where the node subsequently has its clock or timebase counter rolled back to an "earlier" value, will now be unable to emit valid announces that update the path tables of the network.

In such a situation, that destination would potentially become unreachable for `RNS.Transport.PATHFINDER_E` seconds (1 week by default), until its path was finally dropped, and a new announce with the correctly aligned timebase could be admitted onto the network.

Admittedly, that situation is not *supposed* to happen, but if it does anyway, and it probably will at some point, it's worth it to have a mechanism in place to minimize the negative effects.

The most simple solution I can think of for this is something like the following:

- Decide a local "admittance cutoff", ***Ac***. On mostly any system, this can just be decided from the current system time plus a fixed grace time, ***Tg***. Let's say 2 hours for this example. So ***Ac* = *RTC* + *Tg***.
- Define a local "consensus epoch", ***CE*** as the range **[0, *Ac*]**.
- If the announce timebase is not in ***CE***, drop it.

In the cases that a nodes clock is rolled back inadvertently, this minimizes the recovery time to **~*TG***.

For this approach to work consistently even for nodes without an RTC, in the event that their timebase counter is lost, we would have to introduce another metric, which we could call a "network consensus epoch", ***NCE***.

To obtain ***NCE*** on a newly started/joined node without RTC, simply:

- Listen for a while to obtain ***N*** announces from the network, or query the network for a few announces.
- By noting the timebases of the received announces, we learn a "good enough" approximation of the acceptable consensus epoch of remote nodes.
- Calculate a "safe" ***NCE***, for example as:
  - Map an integer distribution of the full range of received timebases
  - Set local **CE** to the minimum value of the top 95th percentile of the integer distribution map

I'd love to hear any ideas and input on that idea as well.

markqvist, 2025-02-17:
By the way @regulad, I highly appreciate you also taking relativistic effects into consideration in exploring this issue :)

regulad, 2025-02-17:
> So far so good. Can anyone come up with faults in, or viable attack vectors against that approach?

Should be all good, provided the timestamp is validated properly and included in the signature on the announce. That was the root problem of this issue, lest we forget. I think the timestamp should _always_ be at least some measure of elapsed time, as if a TCP-style `seq` counter was used it may move much slower than a time-based counter, allowing more time for a replay attack if the same Tg is applied to both counter and timestamp timebases. If all timestamps marched at approximately the same speed, then Tg can be constant between nodes. If, for some nodes, it was a counter, and for others, a clock, then the announce would also have to include an "emitter-specific Tg," which is unnecessary data for one announce. 

> Problem with this of course, is that we now again require some consistency of the clocks across the network, delimited by Tg. It's better than requiring full time sync, and even a Tg of 24 hours should be doable, even in the most bizarrely apocalyptic situations, since anyone able to use a computer at that point, should be also able to at least figure out what day it is. If you have Reticulum, you can probably also find a star chart.

Why would you need to figure out what day it is, if you can just listen to the network's announces to find out what the current NAE-timestamp is? Even in that bizarrely apocalyptic situation, if not a single node in the network has a reliable source of time, the epoch can just be set to an arbitrary "now" time, and every node could just come to a consensus on it. You do still need to protect against 51% attacks slipping the time backwards, though. Any effort to implement time syncing in any way will require a lot of threat modelling against a 51% attack.

> By the way @regulad, I highly appreciate you also taking relativistic effects into consideration in exploring this issue :)

Oh, but of course 😉

closed 2025-02-17 by markqvist

markqvist, 2025-02-17:
> provided the timestamp is validated properly and included in the signature on the announce.

Which it is, of course. The cause of this issue was simply the oversight of forgetting to actually check it against the highest known timebase for the path, in the case of lower hop-count announces arriving.

>  I think the timestamp should always be at least some measure of elapsed time, as if a TCP-style seq counter was used it may move much slower than a time-based counter, allowing more time for a replay attack if the same Tg is applied to both counter and timestamp timebases.

Yes, that is a very good and relevant observation.

> Why would you need to figure out what day it is, if you can just listen to the network's announces to find out what the current NAE-timestamp is? Even in that bizarrely apocalyptic situation, if not [...]

That's correct, but it also creates a situation where everyone *has* to agree on roughly equivalent NAE, which then comes with all of those mentioned considerations. Whether that is done through some network mechanisms, or a bunch of tech-equipped cavemen coming to roughly the same conclusion on what day it is, are just different ways of getting to the same requirement.

But since networks must be able to exist in isolation from each other, and then attain full connectivity if inter-connected at a later point, the network-determined approach could break that, unless carefully considered.

It's something that warrants further thought and consideration.

LinuxinaBit, 2025-02-17:
> Now, this approach does introduce another inconvenience, and that is that an announce which was inadvertently emitted with a "future blob", where the node subsequently has its clock or timebase [...]

The problem with the RTC approach is that it becomes easier to run out of seconds as well.
A counter only counts up once per announce, and it can be stored more permanently, reducing the likelihood of both of these issues happening significantly.

> I think the timestamp should always be at least some measure of elapsed time, as if a TCP-style seq counter was used it may move much slower than a time-based counter, allowing more time for a replay attack if the same Tg is applied to both counter and timestamp timebases.

Oh, that is something I had not considered. This could definitely help mitigate more replay issues if there is a timeout proportional to the hop count after which an announce will not be accepted. Of course this is a much harder thing to implement than a simple check against the last known announce, but it may be viable if implemented correctly?

regulad, 2025-02-17:
>  [...]and it can be stored more permanently[...]

Just store the time you power off at and restore it when you power on if you lack an RTC. The Tg either side of the shutdown window is acceptable room for error, IMO.

> [...]but it may be viable if implemented correctly?

With the new commit, this is basically how it works. The Tg concept and NCE is just a thought experiment for now.

regulad, 2025-02-17:
> That's correct, but it also creates a situation where everyone has to agree on roughly equivalent NAE, which then comes with all of those mentioned considerations.

This is the problem... I sure hope someone comes up with a clever (but not fragile!) way to do this. Maybe we open a discussion about it?

markqvist, 2025-02-17:
Can't hurt!

RFnexus, 2025-02-17:
> Why would you need to figure out what day it is, if you can just listen to the network's announces to find out what the current NAE-timestamp is? Even in that bizarrely apocalyptic situation, if not [...]

"bizarrely apocalyptic situation" is the kind of the thing you can use RNS for :D

Why not just have this on the application level? Some type of "time utility" node (think of something like NIST's [WWV](https://en.wikipedia.org/wiki/WWV_(radio_station))) where all it does is send out the current timestamp as part of announce data, with some identification that "this is a time node, you can establish a link here and try to sync your system clock with it". The interval at which it announces would be the same as for any other destination (6+ hrs). So the point of the timestamp included in announce data would not be to use for synchronization but to compare it to other time nodes on the list based on when it announced and what the included timestamp was. 

The time nodes would be ordered by hop count, with closer ones appearing first to try and minimize latency. This would be a manual process similar to how you can select an LXMF propagation node. 

Idk maybe this is extremely far fetched but I think it would be an interesting solution if you had no other way to get the time.