markqvist/Reticulum #741: bug: any data size will be compressed when auto_compres=False

evgmik, 2025-03-01:
Hi,

The following line has a misplaced `(`, it work as intended when `auto_compress` set to `True`, but if it is `False` then the condition ignores `len(self.uncompressed_data)` because `False and number` produces `False` which is numerically 0. 

Consequently, any data size will be compressed when `auto_compres=False`

https://github.com/markqvist/Reticulum/blob/08751a762af044670c6eb23574a6dca725fd6e0a/RNS/Resource.py#L338

This line should read as 
~~~~
if auto_compress and (len(self.uncompressed_data) <= Resource.AUTO_COMPRESS_MAX_SIZE):
~~~~
note the opening `(` location.

Also probably `auto_compress` should be `self.auto_compress` so the class uses its own variables. Right now it does not matter but in future it might.

markqvist, 2025-03-01:
> The following line has a misplaced ( ...

Are you certain? Does the erroneous behaviour you expect actually occur when running RNS, or is this a "theoretical bug"? See below.

```python
>>> (True and 9 <= 10)
True
>>> (True and 11 <= 10)
False
>>> (False and 9 <= 10)
False
>>> (False and 11 <= 10)
False
```

Are you missing that there is a `len()` call in there?

> Also probably auto_compress should be ...

Why? I don't see any reason for that. The only thing that would serve, as we are in `__init__`, is to add 10 characters to 1 line of code, and 5 characters to 2 other lines of code.

evgmik, 2025-03-01:
Well you are right, it is behaving as it should.

Now I a bit ashamed of myself, I did run some test but I put closing `)` in wrong place at my test time.

As for `self.auto_compress` it was  totally optional suggestion. In well debugged code like yours, it would be unnecessary. I prefer to have a one source of "truth" which are inner fields of the an and thus use `self.` fields down the line. I still adopting to your style, and now I see why you made this choice.

closed 2025-03-02 by markqvist