markqvist/Reticulum #752: announcements are not sent when rnsd is running

acatabc, 2025-03-11:
The `rnid` CLI tool does not send an announcement on the link if `rnsd` is running with reticulum version 0.9.2.
In version 0.9.1 this issue does not occur.

Steps to reproduce:
reticulum config:
```              
[reticulum]                                 
                                 
  enable_transport = False                                                                     
  panic_on_interface_error = No
                                               
                                                                                               
[logging]      
                       
  loglevel = 4                                                                                 

[interfaces]

[[UDP Interface]]
  type = UDPInterface
  interface_enabled = True

  device = <Your Interface>
  port = 4242
```
Step 1:
Open wireshark (or packet capture program of your choice) on the configured interface.

Step 2:
Generate Identity:
``` bash
rnid -g id
```
Step 3:
Start rnsd with the configuration above.
``` bash
rnsd
```

Step 4:
In another shell, send an announcement
``` bash
rnid -i id -a test.test
```
**Expected behavior:**
In reticulum version 0.9.1 you will see the respective UDP announcement packet in the packet capture.

**Observed behavior:**
In version 0.9.2, you will observe that there is no UDP packet sent out over the link in the packet capture.

**Expected behavior:**
In reticulum version 0.9.1 you will see the respective UDP announcement packet in the packet capture.

**System Information**
- Debian 12
- python 3.11.2
- Reticulum version 0.9.2

**Additional context**
If `rnsd` is not running the announcement can be observed on the link.

markqvist, 2025-03-11:
Thank you for the detailed report! I will look into it.

markqvist, 2025-03-12:
Yes, absolutely, that would be very helpful!

commit b0444104 (2025-03-13): Wait for announce timebase tick when announcing via rnid. Fixes #752.
closed 2025-03-13 by markqvist