markqvist/Reticulum #850: License Clarifications

LinuxinaBit, 2025-06-22:
_Note: this was originally supposed to be a PR but I don't think I have the legal right to actually contribute it that way so I am submitting it like this instead. Sorry for any inconvenience._

I know previously I have been critical of the original license change, but beyond compatibility issues I truly have no issues with the actual intention of the license.

That being said, this licence revision would change three specific sections for added clarity and to remove loopholes.
_**Disclaimer: I am not a lawyer; these changes are provided as-is and you bear full legal responsibility for any changes made based on this suggestion.**_

1. Make it version 2; self explanitory.
2. Move the placement of "purposefully" to state "purposefully includes" instead of "purposefully do harm" to move the legal liability from the user to the developer of derivative software.
3. Re-word the machine learning clause for clarity. This may have subtle legal implications, please review carefully.

As always please ask if you have any questions.

```
Reticulum License Version 2

Copyright (c) 2016-2025 Mark Qvist

Permission is hereby granted, free of charge, to any person obtaining a copy
of this software and associated documentation files (the "Software"), to deal
in the Software without restriction, including without limitation the rights
to use, copy, modify, merge, publish, distribute, sublicense, and/or sell
copies of the Software, and to permit persons to whom the Software is
furnished to do so, subject to the following conditions:

- The Software shall not be used in any kind of system which purposefully
  includes amongst its functions the ability to do harm to human beings.

- The Software shall not be used, directly or indirectly, in the training or
  development of any artificial intelligence, language model, or machine
  learning system, nor shall it be included in a training dataset for such a
  model or algorithm.

- The above copyright notice and this permission notice shall be included in
  all copies or substantial portions of the Software.

THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND, EXPRESS OR
IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES OF MERCHANTABILITY,
FITNESS FOR A PARTICULAR PURPOSE AND NONINFRINGEMENT. IN NO EVENT SHALL THE
AUTHORS OR COPYRIGHT HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER
LIABILITY, WHETHER IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING FROM,
OUT OF OR IN CONNECTION WITH THE SOFTWARE OR THE USE OR OTHER DEALINGS IN THE
SOFTWARE.
```

LinuxinaBit, 2025-06-22:
The "loophole" I referred to is that the harm clause originally requied the user of the software to be able to do the harm *on purpose*; if the function was only available by accident (but intentionally included by the developer) it _**could**_ still comply with the license.

_**Disclaimer: I'm still not a lawyer.**_

LinuxinaBit, 2025-06-30:
I went through a lot of waffling on the harm clause.
As you know, it currently states:
`The Software shall not be used in any kind of system which includes amongst its functions the ability to purposefully do harm to human beings.`
It felt... off?
So I first tried moving "purposefully" like so:
`The Software shall not be used in any kind of system which purposefully includes amongst its functions the ability to do harm to human beings.`
But after doing that I realized what was wrong.
The term, as written, relies heavily on the "ability" to cause harm, which doesn't distinguish between intentional and accidental harm.

The "purposefully" was (from what I can tell) intended to make it so that only intentionally harmful features are a license violation, but it reads to me a lot more like it's talking about the end user's intentions rather than the developer's (yeah, English is weird).

Take cars - people run other people over with them [all the time](https://duckduckgo.com/?q=person+drives+into+crowd&t=fpas&ia=news&iar=news).
Would that car then include the ability, for the operator, to purposefully do harm to someone, and would thus make it a license violation to integrate Reticulum into a car?

This is an imaginary exercise, but I assure you that this isn't in bad faith; I really want this license to be written well because hopefully in the future you, Mark, won't be the only one issuing it, and truth be told, I won't take a pinky promise that I won't be held liable from *anyone*.

This also means, however, that me just moving the "purposefully" around (or even adding both of them) isn't going to cut it; it needs a more comprehensive reword:
`The Software shall not be used in any kind of system that is intended by its creator(s) to do harm to human beings.`
There! Intention is now properly invloved and the liability aimed squarely at the developer.
The "used by" does make it tricky; similar word gymnastics can add loopholes that I don't like, so what about:
`The Software shall be neither included in nor used by any kind of system that is intended by its creator(s) to do harm to human beings.`
Seems alright to me... (_**I'm not a lawyer**_)

So TL;DR why make this change?
Because ability does not equal intent, and patching it up to try and mean the same thing leaves very lawyer-able grammar loopholes that are just not ok to have in a software license with potential to be used so widely.

Oh and I like to edit posts a lot; if you're reading this via e-mail, it's probably out of date ;)

markqvist, 2025-07-18:
In about two days, I'll be off for a long break, but before I go, I wanted to carelessly throw around a few ideas regarding this. This will be a lot less structured and thought out than usual, and is basically just a opener for discussion and consideration.

While I currently have absolutely no firm conclusions on this, I've been thinking about the possibility of restricting model training to *only* proprietary models, meaning models that do not provide **all** of the following under an open license, compatible with the Reticulum license itself (or some other defined metric):

- The complete model architecture, and any related source code
- The source code necessary to reproduce all stages of the training environment and process
- The final model weights of any trained model
- Potentially others?

I'll invite anyone who has something to say on this to join in. This might get chaotic, but let chaos ensue.

LinuxinaBit, 2025-07-18:
Discussion is here: #879

LinuxinaBit, 2025-07-29:
It's also worth noting that Beechat has an MIT licensed implementation of Reticulum with AES-256 that is MIT licensed, so this whole license faff is a largely useless endeavor unless you plan on suing them.

LinuxinaBit, 2025-09-27:
I think this blog post from the [Unlicense](https://unlicense.org/) creator that I just found aligns with my thoughts pretty well:

https://ar.to/2010/01/set-your-code-free

A few of the examples are a little extreme in an attempt to prove a point, but I think it makes a lot of sense in spite of that.

**Some choice excerpts:**

> What copyrighting your code really comes down to is this: when you include a license statement in your software project, you are effectively making a threat to the users of your software.

> Now, I don’t know about you, but the least effective way to secure my cooperation is to threaten me. That virtually guarantees that I’ll do the opposite of what you want (even at personal cost to myself), or at the very least that I will choose not to associate with you in the future.
> 
> When I say that copyleft code is contaminated, I very much mean that. It is code with significant legal strings attached, and when using it you had better be very aware of those strings. Again, of course, that’s the point of the [FSF](https://www.fsf.org/)’s strategy, so in case you agree with their missive you will see this as a benefit. I personally don’t, because I have no interest whatsoever in wreaking havoc on people who want to use my code or (heaven forbid) profit from it. I have better things to do.

> It is a stranger thing, even, to believe that slapping a licensing statement on your software will in some way prevent such wonderful “abuse” in the first place. That is simply a form of [magical thinking](https://en.wikipedia.org/wiki/Magical_thinking). Those who would benefit from your software without reciprocity will do so in any case, with or without your implied copyright-sanctioned threats; and regarding my software I say, they’re welcome to knock themselves out. It’s no loss to me.
> 
> Not only copyleft advocates but anybody affixing a licensing statement to open-source software is guilty of either magical thinking or of having an intention to follow up on the implied threat. The only thing your license will prevent, or at least curtail, is the adoption of your software.

I would like to note that because RNS is supposed to be a reference network stack, it seems _even more pointless_ to license it restrictively at all.
I2P knows this; their code is [almost entirely public domain](https://geti2p.net/en/get-involved/develop/licenses) (or at least close to it).

At the risk of sounding completely crazy, you *could* also dedicate RNS to the public domain...
Something worth considering at the very least.

LinuxinaBit, 2025-09-28:
I brought this up in the Matrix channel and got some responses I'd like to save here for future reference:

[image]
[image]

[image]
[image]

[image]
[image]

That last one reminded me of something I'd read a while back, on a different subject, but still very applicable here:

> Here’s what the computer lab donors missed, and it’s a lesson that echoes through decades of development economics research: technology is only as good as the ecosystem that supports it.
> 
> In the 1990s, economists began studying why some technology interventions succeeded while others failed. The pattern was clear: success depended far less on the sophistication of the technology than on whether it fit the local context. A brilliant innovation in Boston might be completely useless in Kampala—not because the people are different, but because the infrastructure, training systems, and cultural context are different.

\- https://kiwix.org/en/the-million-dollar-computer-lab-that-taught-nobody-good-intentions-and-a-better-solutions/

cobraPA, 2025-09-30:
> cobraPA has been unable to legally compile Sideband for a while, and this seems to be the solution going forward: https://github.com/cobraPA/Reticulum-mit https://github.com/cobraPA/LXMF-mit

Well, kind of feels like you've gone on a rant about the license in the last few days...  But seems like Mark is on sabbatical, so doesn't seem productive to keep all that up.  There are no restrictions on 'compiling' anything, so I'd prefer that not used as an argument.  But I do believe the license change is not compatible with the iOS Sideband binary distribution.  Note that 'distribution' is exactly what Mark said he could apply limits to.  As I said a couple other places, I've paused on pulling code with the new license in to Sideband for iOS for now.  Maybe short term, maybe long term.  That is still not decided.  I am working on multiple areas of Reticulum for my own use and plan to continue making the 'beta' of SidebandOS available, but it will be limited to AES128 for now.  I do not personally want to go against Mark's wishes in general, but I do want to use what I have been working on for a number of years up to now.  The MIT repo's I made are for my use while working on SidebandOS and microReticulum/LXMF (and as references for the various RNode targets I maintain or contributed to Mark's repo.)

worst-git-server, 2025-10-02:
why not [WTFPL](https://www.wtfpl.net/)?

joakim, 2025-10-23:
1. Is it really that big of a deal? If the license only applies to the reference implementation in Python, it can't stop other implementations from using another license, can it? Maybe the outcome of this is a Rust implementation with full feature parity licensed under Apache-2.0. That would only boost Reticulum's adoption.

2. I fully support Mark's intentions, and I truly wish it were possible to prevent harmful outcomes of one's technology with a bit of text. But provisions like these are not practically enforcable when you're up against the lawyers of billionaires. Maybe the best one can hope for is to tell the world where you stand ethically and hope that most will respect that. In that spirit, it might be just as effectful to move the two lines from the `LICENSE` to the `README` and make Reticulum's ethical stance loud and clear.

3. If you do change the provisions to make them more likely to hold up in court, there is something called [Responsible AI Licenses](https://www.licenses.ai/). They have a [license generator](https://www.licenses.ai/rail-license-generator) where one can pick and choose legalese provisions like these:

   > You agree not to Use the Artifact or its Derivatives in any of the following ways:
   > - For weaponry or warfare.
   > - For purposes of building or optimizing military weapons or in the service of nuclear proliferation or nuclear weapons technology.
   > - For purposes of military surveillance, including any research or development relating to military surveillance.
   > - In connection with activities that present a risk of death or bodily harm to individuals, including inciting or promoting violence, abuse, or any infliction of bodily harm to an individual or group of individuals.
   > - To generate or disseminate personal identifiable information that can be used to harm an individual or to invade the personal privacy of an individual.
   > - To engage in, promote, incite, or facilitate the harassment, abuse, threatening, or bullying of individuals or groups of individuals.
   > - To generate and/or disseminate malware (including - but not limited to - ransomware) or any other content to be used for the purpose of Harming electronic systems.
  
   Of course they don't have anything regulating AI training, it is a license generator for LLMs after all. But maybe there's something there for the harm provision.

LinuxinaBit, 2025-10-23:
> Is it really that big of a deal? If the license only applies to the reference implementation in Python, it can't stop other implementations from using another license, can it? Maybe the outcome of [...]

I actually agree with this; the hard part is just getting to the point where that happens.

>  I fully support Mark's intentions, and I truly wish it were possible to prevent harmful outcomes of one's technology with a bit of text. But provisions like these are not practically enforcable, especially versus the lawyers of billionaires. Maybe the best one can hope for is to tell the world where you stand ethically and hope that most will respect that. In that spirit, it might be just as effectful to move the two lines from the LICENSE to the README and make Reticulum's ethical stance loud and clear.

That sounds like a really good middle ground, avoiding unintentional legal implications and community roadblocks while still taking a strong moral stance.
I like it a lot 💟

kujeger, 2025-11-03:
Speaking as a pretty copyleft-minded kind of person, would dual-licensing with the current/modified Reticulum license and the AGPL3 be something that would seem reasonable?
That way the MIT-ish (speaking very loosely) reticulum license applies to those that want that, while for example distributions could ship it AGPL3-licensed.

markqvist, 2025-11-04:
Has always been. This has been on the Reticulum website since day one, and back in the early days, before that existed on my own website when the protocol was only described there.

[image]

> HOLY S*** OMG OMG.

Yeah, we're getting very close now. Had to get all the propagation stamp stuff and extra protocol signalling sorted first to be able to implement it.

> If you think it would actually help

Thanks, it's nice of you, but it's not so much writing the text, right now it's more dedicating the 4-5 full days I need to sit down and research it, think about it, research some more, think some, in a seemingly endless loop.